Atmosphere-NX/Atmosphere #2092: Error Code: 2002-4153 (0x207202) wehn trying to launch atmosphere

axcelmo, 2023-04-10:
## Bug Report

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

yes, by trying to launch atmosphere

### Crash Report

 [gist](https://gist.github.com/axcelmo/9a91abfb30ad30544f8c01facf770148)

### System Firmware Version

15.0.1</br>

### Environment?

-  fusee atmosphere-1.5.1-master-
 - Hekate official release version 6.0.2
- partitionbased EmuMMC

Masamune3210, 2023-04-10:
qlaunch crash makes me think custom theme, try cleaning out the sd card folder named the same as the titleid in atmosphere\contents, if the folder exists just delete the entire folder

axcelmo, 2023-04-10:
tried that, there is no folder named with the titleid that I can delete

SciresM, 2023-04-19:
`fs::ResultUnclearedRomRealDataVerificationFailed()` -- emummc is corrupted, back up your data and re-clone.

closed 2023-04-19 by SciresM