Atmosphere-NX/Atmosphere #2094: microSD access error

manop81, 2023-04-14:
## Bug Report

### What's the issue you encountered?
I'm getting an error about microSD card access (error 2016-0247) in Atmosphere. However, Hekate seems to be able to access the microSD without issue. It can identify card information and I was able to write to the microSD card and do a full MMC backup.

Background:
While using Atmosphere with emuMMC, I encountered a crash when I went to change network settings. I tried rebooting and the system got stuck on the second loading screen (after the Atmosphere logo). After several attempts, I tried booting OFW sysNAND. That worked. There was an update downloaded and I let it update. After that, I started getting the microSD access error in OFW.

I went to a backup of my emuMMC and that worked. I was able to do a factory reset and I now I have that emuMMC backed up. Unfortunately, I started seeing the microSD access error in Atmosphere.

In all this, I rebooted many times and tried different microSD cards. I'm not sure exactly when the access error appeared. It might have happened once or twice before I did the factory reset. Now it happens every time.

### How can the issue be reproduced?
Unknown.

### Crash Report
N/A

### System Firmware Version
Atmosphere 1.5.1.

### Environment?
Hekate 6.0.2.
Atmosphere 1.5.1. official.
Clean install without additional modules, kips, or homebrew.
File-based emuMMC.
FAT32 formatted microSD cards.

### Additional context?

I have tried restoring emuMMC backups that I have which were known to be working. I get this error in all cases now, including booting to sysNAND Atmosphere, emuMMC Atmosphere and sysNAND OFW.

manop81, 2023-04-17:
Update:

I have verified with Hekate that there is no problem with the microSD interface on the Switch. I have been able to read and write gigabytes of data without error. I checked all the files on my PC. Hekate UMS gives me full access to the microSD card and everything works.

Under Atmosphere, I still get the error. However, I can run apps if I copy them to the root of the card. It appears that `/switch` and other folders are blocked. From hbmenu, I can't see anything unless I switch to Netloader and then go back.

I was able to run ftpd and within Atmosphere, copy a full system update to the card in `/test`. Note that ftpd gets an I/O error and cannot list the root. I am able to list `/test` but not `/switch`. Furthermore, I was able to install the system update using Daybreak from the files I sent via FTP.

This proves that Atmosphere can get to the microSD card but is being prevented by something else from accessing `/switch` and other files.

SciresM, 2023-04-17:
Delete the `/emummc/whatever/Nintendo` directory. An SD card can only be used with a single console at a time, and a post-factory-reset emummc is "a different console" as NS accounts it.

SciresM, 2023-04-17:
fwiw, this isn't an Atmosphere issue, 2016-0247 is Nintendo's code reporting unexpected result during NS's sd card mount logic. It'll just be because your SD card isn't how ns expects it to be.

SciresM, 2023-04-17:
(Also, this issue tracker is for bugs, not for support, please seek further help on discord).

closed 2023-04-17 by SciresM

manop81, 2023-04-17:
> Delete the `/emummc/whatever/Nintendo` directory. An SD card can only be used with a single console at a time, and a post-factory-reset emummc is "a different console" as NS accounts it.

Thanks. I'll try that but it's weird that my emuMMC backups give me the same error message.

manop81, 2023-04-18:
It didn't work. :(