Atmosphere-NX/Atmosphere #2097: [Feature Request] Method to securely update Atmosphere at runtime

abjugard, 2023-04-18:
## Feature Request

### What feature are you suggesting?
#### Overview:
- a method to securely update Atmosphere at runtime

#### Smaller Details: 
N/A

#### Nature of Request:
- Addition
  - Addition of new functionality to update Atmosphere at runtime

### What component do you feel this would best fit within?
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's custom bootloader.</br>
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

Maybe? I am not completely sure of atmospheres architecture.

### Why would this feature be useful?
For those of us who don't have a convenient way of doing payload injection (I'm using caffeine from sysNAND 4.1.0), updating atmosphere (and managing compatible community sysmodules) is currently a cumbersome process requiring rebooting to something like Hekate to gain access to SD card management, manually decompressing and replacing files, and rebooting.

It would be super helpful if atmosphere could update itself directly from Github releases, with signature checks and everything, and maybe also evict potential incompatible sysmodules after rebooting to the new version.

BeckysFootSlave, 2023-04-18:
Why you don't use AIO Switch Updater???

PoloNX, 2023-04-18:
HI, if it's requested I can make an homebrew to update atmosphere and do a pr. Maybe something like [this](https://github.com/PoloNX/AtmoPackUpdater)

abjugard, 2023-04-18:
> Why you don't use AIO Switch Updater???

I have tried it, it didn't update fusee.bin for me and my boot stopped working. That's partly on me, but it highlights how complicated it can be.

I think the function should be built into atmosphere. It would be so freaking cool if it could be injected into the regular settings in some way, but I doubt that's possible.

Masamune3210, 2023-04-18:
It's been repeated multiple times in these very issues, live updating is not going to be a thing. The only official way to update atmosphere is while not booted into it. Most of the files are actively being used while the console is on anyway, how would you update those?

PoloNX, 2023-04-18:
> It's been repeated multiple times in these very issues, live updating is not going to be a thing. The only official way to update atmosphere is while not booted into it. Most of the files are actively being used while the console is on anyway, how would you update those?

There are two files which are used. Package3 and stratosphere.romfs. So when you download it you add a ".temp" at the end and reboot to one payload which rename to the good format

SciresM, 2023-04-18:
I am really, really personally disinterested in automatic or Atmosphere-applied updates. Feel free to make (or use) a homebrew program for this!

closed 2023-04-18 by SciresM