Atmosphere-NX/Atmosphere #2098: Feature suggestion: Profile-specific game language selection

klauskopec, 2023-04-18:
Feature suggestion: please add profile-specific game language selection. Currently, we can select language per game, but not per profile (person) currently playing the game.

Use case: I play games in English, my kids play the same games in our local language. Currently, we have to switch languages (e.g. Using nx locale switcher) every time we switch who's playing.

DraconicNEO, 2023-04-19:
I think this is a duplicate of #547 to elaborate on that one it was declined since per user configs based on GUID would be too hard to manage.

SciresM, 2023-04-19:
Yeah, pretty much. No good way to specify a user in text.

closed 2023-04-19 by SciresM

DraconicNEO, 2023-04-19:
> Yeah, pretty much. No good way to unambiguously specify a user in text.

Yeah it was also explained in #547

klauskopec, 2023-04-19:
Thanks for clarifying, didn't see the other one.

I guess if you enabled it while suggesting a ui-based tool to make those changes it might work as that tool could likely read the user id. But I understand it's difficult if you want to keep it without any ui.