Atmosphere-NX/Atmosphere #2101: Crash Report

GlorgusToe, 2023-04-20:
## Crash Report

### What's the issue you encountered?

Trying to launch a game that worked fine before is now crashing after installing new update.</br>
Updated Atmosphere.</br>
Just replaced old atmosphere files with new ones.</br>

### How can the issue be reproduced?

Unsure. Likely unique to my modded game.</br>

### Crash Report

https://gist.github.com/GlorgusToe/3bbd377fe50042bcdee2ddac9cac0feb</br>

### System Firmware Version

16.0.2.</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: official release
  - 6.0.3
- Do you have additional kips or sysmodules you're loading: I don't think so.
- Homebrew software installed: ARCadia, Checkpoint, Chiaki, DBI, EdiZon, Goldleaf, NX-Shell, Switch Themes Installer, Moonlight, Breeze, AiO Switch Updater, RetroArch, mGBA, Tinfoil, 90DNS Testing Utility, Dino Run, EasyRPG Player, ftpd classic, ldnmitm cfg, Linkalho, LOVE, LOVE Potion, megazeux, melonDS, nxdumptool, NXMP, pfba, pfbneo, PPSSPP, psnes, and ReverseNX-Tool.
- EmuMMC or SysNAND: SysNAND

### Additional context?

N/A

SciresM, 2023-04-20:
"My smash mod is crashing" isn't an atmosphere issue. Go talk to the mod makers, please.

closed 2023-04-20 by SciresM

GlorgusToe, 2023-04-20:
Sorry, Thought it was an atmosphere issue because the crash report was in the atmosphere folder