Atmosphere-NX/Atmosphere #2103: error code 2162-0002 (0x4a2) program 010000000000001A

Sanswoods, 2023-04-24:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ when i boot with a rcm loader it wont bood to atmosphere ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. 
[01682351011_010000000000001a.log](https://github.com/Atmosphere-NX/Atmosphere/files/11312271/01682351011_010000000000001a.log)
]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

15.0.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version atmosphere hackate fusse and so fort ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ the awoo installer gold leaf appstore gamecard installer breeze themzer nx tinfoul nx shell switch themes installer edizone kosmos toolbox daybreak emuiibo acropolis and quasar]
- EmuMMC or SysNAND:
  - [ i use sysnand i think ]

### Additional context?

- Additional info about your environment:
- [ i had an issue befor where my custom theme did not work i deleted it and the issue was gone but now out off the blue when booting with the rcm loader it wont work ]

NikosApikos, 2023-05-02:
help same error didnt find a fix

HegyiBazsi, 2023-05-07:
Hy similar issue was solved earlier.
https://github.com/Atmosphere-NX/Atmosphere/issues/736

SciresM, 2023-05-07:
@HegyiBazsi your issue is completely unrelated to this. please don't comment if you don't know what you're talking about.

Xammerr98, 2024-09-04:
So is there a solution?

closed 2025-04-03 by SciresM