Atmosphere-NX/Atmosphere #2111: a fatal error occurred when modding TOTK

BanhMi939, 2023-05-03:
I'm trying to mod TOTK with layerFS, but got a error after applying the patch.

### Crash Report

attached below.

### System Firmware Version

16.0.2</br>

### Environment?

- fusèe
- Official release 1.5.2
- tesla menu, salty nx
- SysNAND

[report_00000000af612c55.zip](https://github.com/Atmosphere-NX/Atmosphere/files/11378799/report_00000000af612c55.zip)

only TOTK got this error, other games work fine.

SciresM, 2023-05-03:
Piracy is not supported.

closed 2023-05-03 by SciresM

frimeone, 2023-05-05:
> I'm trying to mod TOTK with layerFS, but got a error after applying the patch.
> 
> ### Crash Report
> attached below.
> 
> ### System Firmware Version
> 16.0.2
> 
> ### Environment?
> * fusèe
> * Official release 1.5.2
> * tesla menu, salty nx
> * SysNAND
> 
> [report_00000000af612c55.zip](https://github.com/Atmosphere-NX/Atmosphere/files/11378799/report_00000000af612c55.zip)
> 
> only TOTK got this error, other games work fine.

the same here

SciresM, 2023-05-05:
@frimeone Piracy is not supported. If you comment again, I will block you from the issue tracker.