Atmosphere-NX/Atmosphere #2112: If hosts/default.txt MITM domain blocking contains lots of lines (duplicates as well?), Atmosphere crashes on boot

AndrewBraunEE, 2023-05-03:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
If I use the following default.txt in Atmosphere's hostfile, I get an explicit crash.
A fatal error occurred when running Atmosphére.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

[ Describe the issue in detail and what you were doing beforehand. ]</br>

[ Did you make any changes related to Atmosphère itself? ]</br>
No, only config change.
[ If so, make sure to include details relating to what exactly you changed. ]</br>
None

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
This issue is always reproducible.
1. Use a template host file and verify that if I use a file with only the first-line (blocking nintendo telemetry server), Atmosphere boots up correctly.
2. Then attempt to use the host file provided, which blocks many of Nintendo's domains in its portfolio.
3. Verify that Atmosphere always crashes.
[default.txt](https://github.com/Atmosphere-NX/Atmosphere/files/11379097/default.txt)

Coredump binaries, logs, thread info dumps available at
https://drive.google.com/drive/folders/1M5BEJJyz-vOgSs90pkKTb_CGlpd5GRnd?usp=sharing

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version
Latest release on Github as of May 2nd, 2023. Believe this is Atmosphere 1.5.2, Hekate 6.0.2

[01571803323_01006a800016e000.log](https://github.com/Atmosphere-NX/Atmosphere/files/11379100/01571803323_01006a800016e000.log)
[01576297047_01007ef00011e000.log](https://github.com/Atmosphere-NX/Atmosphere/files/11379101/01576297047_01007ef00011e000.log)
[01576443778_01007ef00011e000.log](https://github.com/Atmosphere-NX/Atmosphere/files/11379102/01576443778_01007ef00011e000.log)
[01576443779_01007ef00011e000.log](https://github.com/Atmosphere-NX/Atmosphere/files/11379103/01576443779_01007ef00011e000.log)
[01586713925_010021c000b6a000.log](https://github.com/Atmosphere-NX/Atmosphere/files/11379104/01586713925_010021c000b6a000.log)

12.0.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Hekate
- Official release or unofficial build: 
  - Official 1.5.2
- Do you have additional kips or sysmodules you're loading:
- Not that I'm aware of, other than the typical nintendo-switch CFW guides.
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]
  - I am using EMUMMC. I believe partition-based, not sure how to verify. I just do the default in https://nh-server.github.io/switch-guide/extras/updating/

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

AndrewBraunEE, 2023-05-03:
Note: I am not familiar with how Switch Homebrew and its Framework works internally though I am doing Embedded Work. 
My best guess is that at bootup there is some parsing within a daemon or lib of the hosts-file. It likely fails (or succeeds), but some other portion of the bootup sequence fails with this host file.

I do think it **is** something preventable though, but I'll let the Atmosphere developers make this conclusion / add data as applicable ; )

SciresM, 2023-05-03:
Yeah. @AndrewBraunEE This isn't a bug, per se. Atmosphere aborts when the hosts file is too big. The size limit in place is 0x8000 bytes (see the code [here](https://github.com/Atmosphere-NX/Atmosphere/blob/master/stratosphere/ams_mitm/source/dns_mitm/dnsmitm_host_redirection.cpp#L361)).

Our options, when confronted with a too big file are:
* Display a fatal error (crash).
* Do not apply the hosts file.
* Only apply the first 0x8000 bytes of the host file.

In my judgment, both of the latter two options are less preferable than the first -- if a user is relying on some redirections being in place, *invisibly failing* seems to me unacceptable.

So we're left with "crash", which is what we do.

I guess all I can really say is to try to reduce your redirections file to be less than 0x8000 bytes? I'm not *especially* inclined to bump that limit, given the memory constraints in play. If you want to discuss, I'm on Discord (`SciresM#0524`).

closed 2023-05-03 by SciresM

AndrewBraunEE, 2023-05-03:
I think there are more options than are presented, at least, that could be refactored into (1).
(2) and (3) are just bad UX so I agree with you there.

Two things I would like to ask:
1. Is there a specific reason why not to use the heap here for dynamic allocation? File can have many different number of bytes.
Is your team just suspecting / avoiding fragmentation as much as possible?

2. Display a fatal error (crash)
I agree that this should be the general approach, but I don't think it's 100% the right implementation.
Right now it seems like dnsmitm_host_redirection raises SIGABRT. When this process raises SIGABRT, I assume you have a signal handler that displays portions of the core to the display before further raising SIGABRT to the system (which would generate coredump).

Assuming dnsmitm_host_redirection is in a separated process, you could implement IPC to the DisplayController / Module to display a message _and then crash_ so that the user knows what exactly they goofed up. I would assume this would require reworking some of the current signal handling currently -- as other SIGABRTS should have the same behavior as is currently.

TLDR: User inputs should be sanitized. It seems like you're doing this already. Bad input should be displayed to the display so that user knows how they goofed. If this weren't Atmosphere / on boot, I would argue that this error message should be **non-fatal**. However since this is on-boot into Atmosphere, I agree with treating it as fatal. But UI still needs adjustment.

SciresM, 2023-05-03:
1) That does use heap. You can see the code calls malloc. We are *severely* memory constrained, and the heaps are the minimum size they can be to guarantee that all released games can launch with mods active without crashing. 

However, note that "heap" is not arbitrarily sized. System modules specify how much """heap""" they use at compile/build-time. Malloc is implemented by allocating out of a fixed-size buffer in the .bss segment.

2. It is not possible to display messages from sysmodules, pretty much literally at all, and also cannot take user input in response to failures like this. 

Also, signals don't exist on Horizon, we're not raising SIGABRT since SIGABRT doesn't exist. We data abort, which invokes the fatal error handler, which reboots into a payload that displays the error + saves details to SD (because the SD card is not reliably available in fatal-error context, and in fact the module that does dns.mitm redirection *also* does SD card interception, so this is necessary in this case in particular).

SciresM, 2023-05-03:
fwiw, there are also many other things in atmosphere I would do differently if displaying things to the screen from a system module context (and ideally prompting the user for input) was reasonably feasible.

And, for your reference, 0x8000 is 2-4x as much heap as many of atmosphere's sysmodules have in their entirety :)

SciresM, 2023-05-03:
Also, to be specific:

> Right now it seems like dnsmitm_host_redirection raises SIGABRT. When this process raises SIGABRT, I assume you have a signal handler that displays portions of the core to the display before further raising SIGABRT to the system (which would generate coredump).

The process which implements DNS redirection (`ams.mitm`) calls an Abort() function which intentionally data aborts by writing a magic value to an invalid address.

The kernel will then make the thread execute the abort exception vector; ams.mitm will then collect data about the crash (including a register dump), and format it into an atmosphere fatal error binary (these are the report_*.bin) contents under `atmosphere/fatal_errors`.

ams.mitm will then make a custom system call to invoke the fatal error handler, which will write the report data to some memory (SRAM) that doesn't get cleared on reboot, and perform a reboot to a payload that dumps the report data to the SD card + displays on the screen.

We are severely constrained on memory that can be passed post-reboot; The reports are like 0x500 bytes, which is close to the reasonable limit (0x800).

Also, on newer devices (Mariko), instead of rebooting, I just hijack the entire application processor (killing all but one cpu core) and display directly from TrustZone. This is possible because the newer SoCs have an extra 0x30000 of TZRAM for me to put that code in (there is less than 0x800 of free TZRAM on older devices).

AndrewBraunEE, 2023-05-03:
Sweet, thanks for the discussion SciresM.
I will admit I only skimmed the region-of-code you sent so I was just going off of immediate face-value, but this gives more context on why the design is as-is. When you mentioned that Atmosphere is already memory constrained I figured this was in the order of 10s of MB or 100s of MB, but clearly that is not the case -- the order is extremely smaller.
Having this information now, it's clear why the current implementation is as is right now.

SciresM, 2023-05-03:
Yeah, I'd kill for 10s of MBs to be available :)

Anyway, as I mentioned, if you're interested in asking more questions/if you have other crashes to debug in the future, you're welcome to reach out on discord!