Atmosphere-NX/Atmosphere #2113: Joycon disconnecting / intermitent connection

rcocchiararo, 2023-05-03:
Hi

I am new to this side of the switch world.
I own an original model from when it launched, and aside from having to place "conductive foam" in one of the original joycons to have it work perfectly, i never had connection issues.

Now, after making the emunand, the pro contoller still works fine, but both pair of joycons disconnect if thate is a person between them and the switch, and they work pretty bad unless held in the open air.

If i revert to the "stock mode" they work fine again.

I am running 16.0.2, all my controllers updated, and AMS 1.5.2.

I tried enabling and disabling the AMS usb 3.0 patch, same result.

My problem is only present in the emunand enviroment.

Google presented me no similar issues, just one guy in a forum that was already running AMS told me that he started having similar issues after 16.0.2 but never confirmed if on AMS or stock mode.

I first noticed issues while playing, but then learned to test them with the stick calibration option.

Any help would be appreciated.

Masamune3210, 2023-05-03:
Don't use packs. They can cause all sorts of issues. Download Hekate and atmosphere from their respective repos, remove the current contents of the SD card and copy the downloaded archive's files to where they go on the card and retest.

rcocchiararo, 2023-05-03:
Will Do.
I just tested that it has mission controll by default (tested nvidia shield controller(2015) and it works perfectly :P )

rcocchiararo, 2023-05-03:
Ok, deleted everything (excep NINTENDO and emunand folders).
Droped hekate + atmosphere clean from the download sites (some step missing cause i could not boot except choosing the payload, but ok for testing).

Problem persists.

If i lay my hands on my LAP for playing with the joycons, connection is CHOPPY.

If i raise my legs and they interrupt line of sight to the switch, connection is terrible in general and drops on the colored joycon pair, on the right one (that pair was bought a few weeks after the console)

SciresM, 2023-05-03:
Please post a screenshot of atmosphere/contents?

Genuinely, I have no idea what could cause this. Atmosphere doesn't by default interact with anything that impacts radio/Bluetooth stuff. I can't reproduce locally using joy-cons in emunand.

SciresM, 2023-05-03:
(Also show that you have no custom config in atmosphere/config?)

rcocchiararo, 2023-05-03:
Might be a lost cause then :(
I can more or less manage with the pro controller and the nvidia one. 
The joycons have to be used in a "special way" to have them work ok, or in "legal mode".

Here are the screenshots tho:

[image]
[image]
[image]

there was no content folder when i extracted the .zip (i already it, but i swar y just pasted that into the clean sdcard).

On hekate's side:
[image]

That, plus i pasted hekates and AMS paylods to the payloads folder.

I noticed that both the pack i had used and a friend had a file called "patches.ini" inside the bootloader folder. 
I tested without that.

Masamune3210, 2023-05-05:
I mean, if you really want to test it and have a nand backup, you could always take a backup of the current system so that you can revert and just reflash the nand backup and update to 16.0.1 to test your theory.

rcocchiararo, 2023-05-06:
I have a nand backup of 16.0.2 only.
If i can go back to 16.0.1 in any way that wont have negative consecuenses i can look into it.

alanlcarlos, 2023-05-11:
I also have a problem with disconnecting the controls, the right joycon disconnects if I cover it all with my hand and point it to the ground, I tested it on another SD card with atmosphere 1.4.1 and FW 15 and the problem did not happen, so I updated to Atmosphere 15 and the problem has returned.

rcocchiararo, 2023-05-11:
I feel less alone now :P 

All my controlles seem to have less "connection power" (in docked mode of course) in AMS.
Having a few mobile devices/tablet near the switch makes even the pro controller lag (never disconnect tho).

Trilkin, 2023-05-12:
I'm having this exact issue. Launch Switch, running 16.0.3 on both sysnand and emunand. Using AMS 1.5.3. sysnand has no issues. emunand has big connectivity issues with anything Bluetooth for me.

rcocchiararo, 2023-05-12:
> 

So, external bluetooth devices are fine, only internal bluetooth suffers.

Joycons have historically weak signal, so they suffer the most.

Trilkin, 2023-05-14:
Is there any more information we can provide to help troubleshoot this? I don't have a custom config for Atmosphere either. Only Exosphere stuff. Everything is basically as 'clean' and minimal as it can get.

SciresM, 2023-05-14:
Like I said, I genuinely have no idea what could cause this. We simply do not interact with the bluetooth/radio stuff.

Also, I can't reproduce, despite some effort.

Trilkin, 2023-05-14:
So I don't know if it's a coincidence or not because I wasn't smart and didn't check before I did it, but it actually might be a sysclk issue. I keep forgetting that it isn't a standard AMS component so I didn't initially consider it. I haven't had the issue since removing it, but I can't tell anyone for certain whether that was the culprit. Something to potentially look at for other people having this issue, though.

SciresM, 2023-05-14:
@rcocchiararo Actually looking at your screenshots again, you didn't post a screenshot of sd:/atmosphere/contents. You posted a screenshot of the atmosphere zip, which is different. Share an actual screenshot of sd:/atmosphere/contents?

rcocchiararo, 2023-05-14:
> @rcocchiararo Actually looking at your screenshots again, you didn't post a screenshot of sd:/atmosphere/contents. You posted a screenshot of the atmosphere zip, which is different. Share an actual screenshot of sd:/atmosphere/contents?

I know, i took it before putting it in my "test SD card" (i have my working 512gb one and tried it on my 128gb original one).

Will now test what @Trilkin mentioned about sysclk, i had it in my switch folder in both sdcards, never used it tho)

rcocchiararo, 2023-05-14:
Here is my current working SDcard:

[image]

[image]

[image]

I put mission control back in, because the nvidia switch controller acts like a pro controller and works, allowing me to at least confortably play with 2 non disconecting controllers.

I was wrong about sys-clk, i had the manager app, but i never added the rest, so it was not doing anything.

rcocchiararo, 2023-05-22:
> for me, The issue is about using joycon via bluetooth in both sysnand and emunand. see: #1705

seems like a really different issue to me.

My/our problem happens only with the AMS enviroment, and its not like i cant use the joycons, they just need to be closer to the console and with perfect line of sight.

I reboot to the original firmware (Same version as in AMS) and they work like they used to..

Next weekend i will update to 16.0.3 and testo a clean (again) AMS enviroment just created with nothing again. (i did that in 16.0.2 but failed to take screenshots).

bakugo, 2023-05-23:
Reproduced this issue on a freshly hardmodded oled with an empty contents folder so I can't see how it could be anything other than an atmosphere issue.

Only happens while docked, and interestingly enough, it seems the issue gets way worse when I have a LAN adapter (TP-Link UE306) connected to the USB port inside the dock, it makes joycons borderline unusable 2m away from the console (still happens without it but doesn't feel nearly as bad) so this is probably USB related. Dunno if it's important but the ethernet cable was disconnected at the other end while I was testing (I usually disconnect it while using ams because I'm paranoid).

SciresM, 2023-05-23:
"can't be anything other than an atmosphere issue" I mean, it could be an issue with environment and/or hardware and/or hardmod.

I have still never reproduced this issue myself even once, despite many attempts.

bakugo, 2023-05-23:
Again, the issue doesn't happen without ams running (I play splatoon online, I would absolutely notice if my controller lagged during that) and it happens on both a v1 switch and a brand new oled so it can't be a hardware difference. Like I said it could be some sort of USB interference, USB3 is known to cause issues with bluetooth.

SciresM, 2023-05-23:
Yeah, I'm mostly just saying I do not believe there is anything I can actually do.

rcocchiararo, 2023-05-23:
It should be an AMS thing since booting into OFW "fixes" the issue and going back to AMS makes it appear again.

It is not a NO signal or disconection issue, its like signal streanght is weaker (and since joycons have weaker signal already they work way too bad).

I too had a usb3 ethernet adapter plugged in.

Will test later today if something changes without it.

Also undocking the console.

The easiest way to test it is to go to the analog stick test option in the console configuration.

There, if you are 2 to 3 meters away and raise your legs or put stuff between the control and the console, you can see how the stick movement starts jumping or becomes erratic, and if you keep the interference going joycons tend to disconnect (specially the right one).

rcocchiararo, 2023-05-23:
> Reproduced this issue on a freshly hardmodded oled with an empty contents folder so I can't see how it could be anything other than an atmosphere issue.
>
> Only happens while docked, and interestingly [...]

I love you.

My original, first batch switch (bought on launch on amazon spain) had from day one a usb 3.0 ethernet adapter plugged into the blue connector on the dock.

its a TrendNet TU3-ETG/EUNL.

Removing it fixes this.

I guess AMS has some issue with USB 3.0 stuff plugged even if the AMS usb 3.0 patch is disabled?

In any case, i don't really need it since i never played any online stuff and i can enable 5.0 GHZ wifi  if i do need internet.

SciresM, 2023-05-23:
@rcocchiararo To check: you're saying that the issue occurs with some USB3 adapter plugged in, and you are booting via fusee (not hekate), and you have no atmosphere/config/system_settings.ini? 

I have usb3 disabled, obviously, and I know that it being disabled works fine booting from fusee. Can you post a screenshot of your atmosphere/config/system_settings.ini, if you have one, and also of whatever your hekate boot config is if booting via hekate?

rcocchiararo, 2023-05-24:
> @rcocchiararo To check: you're saying that the issue occurs with some USB3 adapter plugged in, and you are booting via fusee (not hekate), and you have no atmosphere/config/system_settings.ini?
>
> I [...]

The issue went away when I removed muy USB 3.0 ethernet adapter (back in the day there was no 5.0 GHz wifi in Argentina and the switch sucked with 2.4ghz wifi, so I had that).

Since it was 3.0 I had it plugged in the 3.0 blue port on the dock.

I have no settings in (I created one back then to try to enable USB 3.0 to test file transfer speeds).

I think I booted from hekate (never checked/touched Hekate's files).

Will try booting directly with fusee tomorrow.

SciresM, 2023-05-24:
Please post a screenshot to verify no settings, when you test booting with fusee.

akfc58, 2023-05-24:
> It should be an AMS thing since booting into OFW "fixes" the issue and going back to AMS makes it appear again.
>
> It is not a NO signal or disconection issue, its like signal streanght is weaker (and [...]

what if you go to OFW settings controller and forget all joycons by hitting X, and do that again in AMS, and just test connection only in AMS(not boot OFW at all). Would this work for you?

Because I just had the exact same weird behavior like yours when I was playing with my friends back in the days, and I indeed had gone to OFW and uploaded the OFW before that, then the issue appeared. My theory is that the system backed up something about your joycons, and changing firmware makes conflicts, leading to buggy bluetooth connection.

Masamune3210, 2023-05-24:
No chance, the nands are separate by intent, anything that could have been "backed up" would likely cause issues for both even if there was some way for them to conflict,  not just the in the emunand.

rcocchiararo, 2023-05-24:
> what if you go to OFW settings controller and forget all joycons by hitting X, and do that again in AMS, and just test connection only in AMS(not boot OFW at all). Would this work for you?
>
> Because [...]

Tried that long ago, was not related.

The issue was the usb 3.0 network adapter (somehow).

rcocchiararo, 2023-05-24:
> Please post a screenshot to verify no settings, when you test booting with fusee.

[image]

Since i did not know what hekate file to take a pic of, here is everything i could find:

[image]
[image]
[image]

will now test direct boot from tegrarcmGUI to fusee

rcocchiararo, 2023-05-24:
Booting from fusee makes no difference, issue is there, I unplug the adapter and it works fine

andrewmiskell, 2023-05-28:
I seem to be having the same issue, right joy con disconnects from the switch after a few minutes. I do have an ethernet adapter connected and disconnecting it seems to solve the problem for me.

FW: 16.0.3
AMS: 1.5.3(S)

carmona, 2023-05-28:
I've got the same issue as well, but using sysnand and nothing else is connected in the usb ports. Tried disabling MissionControl and sys-con to no avail, so now I just use an xbox wired controller instead of the joycons.

Model: V1 (2017, softmod)
FW: 16.0.1
AMS: 1.5.1(S)

mgumaraesm, 2023-05-30:
I noticed exactly or mesmo erro starting from 16.0.3, antes nao havia percebido nenhum problema.  To investigate this problem of anomalies with the joy of using ams emunand, I would find this discussion and simply solve my problem after disconnecting the ethernet adapter that was on the usb3 port, my device is a v1.  Obligated to all by isso.
 Other than that, the need to report is that from the 16.0.0 you notice that the adapter I just unplugged the appliance, turned on, never turned on the lights indicative of the operation and never worked automatically as before.  To function, you specified to disconnect the adapter from USB 3.0 port and connect them followed to turn it on
 at rede.
 In conclusion, however, that the AMS is in some way interfering with the operation of the docking plate in the way that you plug in the bluetooth operation to connect with the joycons.
 v1 mod
 Erista
 firmware 16.0.3
 customFirmware16.0.3--6
 AMS15.0.4

furkandeger, 2023-06-16:
I have the exact same problem with a V1 switch, on docked mode. I'm also using a usb ethernet adapter that is connected to the rear USB 3.0 port of the dock. After reading through this post, I've noticed the problem goes away if:

1) I use the ofw
2) I use the cfw but remove the ethernet adapter
3) I use the cfw but plug the ethernet adapter into one of the side USB 2.0 ports of the dock

My switch is an unpatched V1 

Firmware 16.0.3
AMS 1.5.2

I have USB 3.0 disabled

akaharaimori, 2023-06-23:
I have V1 erista(Unpatched) and got same issue only when switch is docked. But in my case deleting sys-clk sysmodule fixed this issue(delete /atmosphere/contents/00FF0000636C6BFF).
Sys-clk sysmodule is loaded default in many packs so even if you didn't do anything about sys-clk this might help you.
Firmware 16.0.1
AMS 1.5.1
USB 3.0 disabled

rexquando1, 2023-09-14:
I also have this issue. I don't quite recall when it started, at least 6 months ago. I chalked it up to it being the Switch being the Switch. Current setup is
V1 Erista
16.1.0
AMS 1.5.5
SysNAND - never touched emuNAND
No piracy, no sys modules, nothing custom setup. The only homebrew I have are the apps that come with AMS Checkpoint, and Retroarch for my precious Pokemon. 
Both Joycons will drop inputs I'd say 10% of the time. Ranging 1-15 seconds of "dead time" 
Right Joycon is worse than the left one. 
Right Joycon will fully disconnect from time to time. 
This is playing in the dock and tabletop, no matter how I hold them. 
The issue does NOT happen on OFW

jvguye, 2023-10-24:
> I have V1 erista(Unpatched) and got same issue only when switch is docked. But in my case deleting sys-clk sysmodule fixed this issue(delete /atmosphere/contents/00FF0000636C6BFF). Sys-clk sysmodule [...]

This fixed it! Thank you so much! I've had this problem for half a year now and it was so frustrating when friends came over and the signal was doodoo. This completely fixed it (deleting this folder + removing the Ethernet adapter from the 3.0 port [haven't tested it on the 2.0 one though, but I don't use only much anyway on the OFW]). Please let me buy you a beer :)

Masamune3210, 2023-10-24:
Congratulations, you just stumbled onto the number one reason why those all-in-one packs are disencouraged.

jvguye, 2023-10-24:
> Congratulations, you just stumbled onto the number one reason why those all-in-one packs are disencouraged.

Tbf the packs do help us noob folks in the jailbreak process, and they're the ones that people ask us to use in the tutorials and whatnot. But I feel ya

jvguye, 2023-10-24:
Also, what is this sys-clk (content folder 00FF0000636C6BFF) for anyway?

BeckysFootSlave, 2023-10-24:
We (SwitchBros) also provide one of those AIO Packs, but with the difference that all of our three installers (two bat files for Windows (NERD and NOOB) and a TegraExplorer Script) let the User choose if he wants to install it!

Sadly it is a german AIO Pack only!

By default it isn't installed!

@jvguye all system modules have their own folder in atmosphere with their specific files in it!
So atmosphere knows what to do and like any title on the Switch they have this unique 16 digit TitleID!

akfc58, 2023-10-24:
> > I have V1 erista(Unpatched) and got same issue only when switch is docked. But in my case deleting sys-clk sysmodule fixed this issue(delete /atmosphere/contents/00FF0000636C6BFF). Sys-clk sysmodule is loaded default in many packs so even if you didn't do anything about sys-clk this might help you. Firmware 16.0.1 AMS 1.5.1 USB 3.0 disabled
> 
> This fixed it! Thank you so much! I've had this problem for half a year now and it was so frustrating when friends came over and the signal was doodoo. This completely fixed it (deleting this folder + removing the Ethernet adapter from the 3.0 port [haven't tested it on the 2.0 one though, but I don't use only much anyway on the OFW]). Please let me buy you a beer :)
> 
> Edit: for those of you that, like me, didn't know what sys-clk was: https://github.com/retronx-team/sys-clk

omg, is this life long problem going to an end?! congratulations

jvguye, 2023-10-24:
> omg, is this life long problem going to an end?! congratulations

Thanks! I was so happy yesterday I couldn't believe it. I really hope this is the definitive fix for other people with this problem ❤️

akfc58, 2023-10-25:
> Thanks! I was so happy yesterday I couldn't believe it. I really hope this is the definitive fix for other people with this problem ❤️

Would deleting of that module cause any functionality problem?

jvguye, 2023-10-25:
> Would deleting of that module cause any functionality problem?

As far as I checked, no, as it's an overclock module.

SciresM, 2023-10-25:
I'm going to lock this issue. If the op wants it unlocked, they can dm me on discord. It's generating too much noise.

closed 2025-04-03 by SciresM