Atmosphere-NX/Atmosphere #2115: fatal error titleID 010041544D530000 error DABRT (0x101)

frimeone, 2023-05-05:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

i put a text mod on the content file with the right ID but I had a fatal error, the game is Tears of the kingdom</br>
[ Did you make any changes related to Atmosphère itself? No change ]</br>
</br>

### How can the issue be reproduced?

[ * ]</br>
put any file on teh contents files of atmosphere with totk game</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

16.0.2</br>
</br>
last atmosphere 1.5.2</br>
</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: official
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: sysnand
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2023-05-05:
Piracy is not supported.

closed 2023-05-05 by SciresM

frimeone, 2023-05-05:
piracy? i want just try a mod

SciresM, 2023-05-05:
"totk game".

That game is not released for another week. You are pirating it. Piracy is not supported.

frimeone, 2023-05-05:
ok thank you but i just want to know if the error is a wrong manip from me. Just that

SciresM, 2023-05-05:
Not supported means not supported.

frimeone, 2023-05-05:
thank in advance