Atmosphere-NX/Atmosphere #2116: switch gets slow

Litri23, 2023-05-06:
hello! I've been using atmosphere for a long time, for a while I've been experiencing that after a day or two the switch becomes slow. If I turn off and launch payload again, it stays loading in the switch symbol for a long time. I have discovered that the problem is in the erpt_report folder, after deleting that folder (165 megabytes) everything returns to normal, working perfectly.

SciresM, 2023-05-06:
Yeah, that makes sense.

closed 2023-05-06 by SciresM

Litri23, 2023-05-06:
Is there a way to avoid this without having to remove the micro sd every time to delete that folder?

SciresM, 2023-05-06:
Well, you can mount the uSD via hekate, so you don't need to remove it. But this isn't actually a bug. You presumably have some sysmodule or config or bad setup that causes tons of reports to be generated, if you're interested you could probably solve that somehow, but this issue tracker isn't the place for it.

Litri23, 2023-05-06:
Thank you very much for your prompt response. I don't understand English or github very well, so I don't know where is the place to mention this. I am using the translator to be able to communicate with you. It would be of great help to me. I have the switch updated to 16.0.2 with atmosphere and hekate downloaded directly from their github repositories. I haven't touched any settings. the only thing that when I launch hekate for the first time after pasting the files I have to press on Fix archive bit, so that it loads atmosphere thanks for your answer. I stay in your hands