Atmosphere-NX/Atmosphere #2117: Atmosphere Fatal Error - Program ID: 524850534f53454d

Aminity0, 2023-05-07:
**What's the issue you encountered?**

I'm on 16.0.2, with Atmosphere 1.5.2, After using latest Fusee.bin, I got

"A fatal error occured when running Atmosphere."
Program ID: 524850534f53454d
Error Desc: Kernel Panic (0xf00)

Report saved to /atmosphere/fatal_errors/report_0000000000a8e66d.bin"

Switch failed to boot after updating to latest firmware version from 14.0, CFW failed to boot. tried reflashing nand,  and then got this issue, latest atmosphere still fails to boot now.

**System Firmware Version**
16.0.2

**Environment?**

    Atmosphere 1.5.2 with its fusee.bin

**Additional context?**

N/A
[report_0000000000a8e66d.zip](https://github.com/Atmosphere-NX/Atmosphere/files/11413915/report_0000000000a8e66d.zip)

SciresM, 2023-05-07:
Yeah, this is the usual end-of-RAM hardware issue that generates mesosphere fatal error reports.

Can you confirm whether your console can successfully boot stock without atmosphere?

Aminity0, 2023-05-07:
> Yeah, this is the usual end-of-RAM hardware issue that generates mesosphere fatal error reports.
> 
> Can you confirm whether your console can successfully boot stock without atmosphere?

It cannot

SciresM, 2023-05-07:
Yeah, this is a hardware problem. Gonna close because hardware issue, but the long and short of it is that the end of your RAM is no longer reliable.

Aminity0, 2023-05-07:
> Yeah, this is a hardware problem. Gonna close because hardware issue, but the long and short of it is that the end of your RAM is no longer reliable.

Alright welp, that's unfortunate, thanks for the help!

closed 2023-05-07 by Aminity0