Atmosphere-NX/Atmosphere #2119: gcc-13: ams.mitm crashes when launching Fire Emblem: Engage with romfs mods

SciresM, 2023-05-07:
gcc 13 (well, okay, it's really updated newlib, but whatever) increases the malloc overhead size slightly. This increase multiplied by the 200000 files in Fire Emblem: Engage's romfs means that we are once again running out of memory when trying to build the romfs for mods.

I think a further increase to the system pool is untenable, and unfair -- I think I'm just going to have to bite the bullet and implement dynamic pool stealing to take memory away from the game to use for romfs mods. Ideally, this behavior will be whitelisted and only on for games that need it (Animal Crossing, Fire Emblem: Engage). Maybe we can reduce the non-dynamic heap size once this is done, too, and get some benefit out of it for saner use cases.

This will involve, probably:
* Adding a new interface method to e.g. pm:shell for ams.mitm to let pm know when it needs to steal memory from the application pool;
* A way for pm and ams.mitm to synchronize memory-release on application exit, since the usual defer-unload strategy isn't really desirable, if we're going to be doing dynamic stealing,
* Adding logic for pm to do system-resource boost based both on the value-from-ns and however-much-ams.mitm-is-stealing.
* Adding logic to ams.mitm to use dynamically stolen memory for romfs allocs. Easy enough to do SetHeapSize() to dynamically create and release heap based on stolen memory, but I can already tell that lifetime issues are going to be *really* annoying, because allocations can persist past building. 

Anyway, absolutely fuck me, I can't believe I've been burned by fire emblem: engage three times now, christ.

AMSNX, 2023-05-08:
Legend of Zelda Tears of The Kingdom crashs too with mod, a brazilian group is making a translation for this game to portuguese,  atmo crash when open the game

Slluxx, 2023-05-11:
Will this solution be game-independent? Because there are more games that have this issue due to a massive romfs structure.

SciresM, 2023-05-11:
@Slluxx No, it will not be game-indepedent. Game independent solution is not possible.

I will be relying on people reporting games that are broken to me.

SciresM, 2023-05-11:
@Slluxx What other games are known to be broken?

Slluxx, 2023-05-11:
@SciresM Some people that got early copies of TOTK report the same issue. I know its not officially released and one can not expect you to do anything for it. Its just an example. I also know of one more game (released a **long** time ago) but right now the name wont pop into my head. I'll tell you if i remember.

SciresM, 2023-05-11:
I'd say piracy isn't supported, but TOTK is in fact out in Japan, so meh. Yeah, I'm aware.

masagrator, 2023-05-12:
@Slluxx  if you mean Animal Crossing, then it's already listed in source code.

dnoky, 2023-05-12:
So can we expect any updates? about TOTK?

grayVI, 2023-05-12:
Is this issue also reflected in previous versions of the system?

SciresM, 2023-05-12:
@grayVI See the changelog for Atmosphere 1.4.1. I do not know what you're asking.

SciresM, 2023-05-14:
I can already feel that this is going to break in the future. c'est la vie.

closed 2023-05-14 by SciresM