Atmosphere-NX/Atmosphere #2123: Atmosphere crashing with Amiibos

BigBoss320, 2023-05-12:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Atmosphere is crashing when wanting to scan an Amiibo in certain apps.

### How can the issue be reproduced?

Connect a controller without the Amiibo scanning feature, start an app that includes Amiibos, like Zelda:Breath of the Wild, start the scan.
Atmosphere gives the followed error code : 2002-0001 (0x202)

### Crash Report

https://gist.github.com/BigBoss320/6c39744531907b64d619dc353496982c

### System Firmware Version

16.0.3</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - Official release version 1.5.3
- Do you have additional kips or sysmodules you're loading: I just completely redo the SD to see if it was the Sysmodules. So, no.
- Homebrew software installed: Awoo Installer, FTPD Pro
- EmuMMC or SysNAND:
  - SysNand

### Additional context?

- Additional info about your environment:
- No

Masamune3210, 2023-05-12:
Piracy is not supported.

BigBoss320, 2023-05-12:
Piracy??

Masamune3210, 2023-05-13:
[image]

BigBoss320, 2023-05-13:
That's not because of that! I have the physical game

Masamune3210, 2023-05-13:
Doesn't matter, Atmosphere considers anything that involves installing NSP's to be piracy and will not offer support.

BigBoss320, 2023-05-13:
And so? Will removing the hemobrew solve the problem?

Masamune3210, 2023-05-13:
Really struggling with the whole "No Support" part of this equation, arent you

closed 2023-05-13 by SciresM