Atmosphere-NX/Atmosphere #2126: USB File Transfer not working in Atmosphere 1.5.3

laurekamalandua, 2023-05-14:
Since updating to 1.5.3 my device is not recognized (either PC or MAC) using the built in haze.nro application. I tested MTP transfers with the same device/cable in Android and have no issues connecting this way. Can someone verify whether this piece of functionality was tested in the latest release and verified to be working?

**Device details** 

| Component | Information |
| -- | -- |
| Atmosphere Version | 1.5.3 |
| OFW version | 16.0.3 |

SciresM, 2023-05-14:
I did in fact test it and have used it on both 1.5.3 (and the 1.5.4 pre-release).

SciresM, 2023-05-14:
You also, notably, didn't fill out the issue template. Please fill out the issue template.

shingoxray, 2023-05-15:
Same problem on macOS. Android File Transfer and OpenMTP not recognizing the device. But on Windows it is OK.

AuroraWright, 2023-05-16:
Works here with both OpenMTP and go-mtpfs on macOS

laurekamalandua, 2023-05-21:
It was resolved with a clean reinstall.

closed 2023-05-21 by laurekamalandua

shingoxray, 2023-05-22:
> 

Reinstall the ATM?

sourcec0de, 2024-04-18:
Uninstalling Google Drive did the trick for me.

lcn918, 2024-08-24:
Same issue when using a newer version (1.7.1). Quitting the Google Drive app then reconnect Switch worked for me. The Android File Transfer app recognized Switch correctly.