Atmosphere-NX/Atmosphere #2127: TotK mod error

tsurugikengo, 2023-05-14:
When I install the mod and start the game, it crashes and this screen appears.
[image]

SciresM, 2023-05-14:
Please upload the report file, and fill out the issue templates.

SciresM, 2023-05-14:
Also, provide details on the mod you're using?

SciresM, 2023-05-14:
...also, are you using the 1.5.4 prerelease? "Fatal error" is expected on 1.5.3.

tsurugikengo, 2023-05-14:
I use 1.5.4
mod is Edited weapon parameters by myself

SciresM, 2023-05-14:
Please upload the error report binary. And the mod, I guess.

SciresM, 2023-05-14:
And a screenshot of atmosphere/contents in general?

tsurugikengo, 2023-05-14:
Give Up.
thank you very much.

SciresM, 2023-05-14:
...okay, I'll close this, I can't help you unless you upload the report binary.

closed 2023-05-14 by SciresM