Atmosphere-NX/Atmosphere #2129: dmnt_cheat_vm Documentation/code mismatch

DKingAlpha, 2023-05-15:
I was writing assembler and disassembler for dmnt cheat codes.
By really reading the doc and code I found an inconsistency in code 0xA.

In  [dmnt_cheat_vm.cpp: Ln.1065](https://github.com/Atmosphere-NX/Atmosphere/blob/3cb54e2b4bbd874da83f4ab673cb54c9f0c34e32/stratosphere/dmnt/source/cheat/impl/dmnt_cheat_vm.cpp#LL1066C39-L1066C39) of main branch
```
case CheatVmOpcodeType_StoreRegisterToAddress:
    // ...............
    case StoreRegisterOffsetType_MemImmReg:
        dst_address = GetCheatProcessAddress(metadata, cur_opcode.str_register.mem_type, 
            m_registers[cur_opcode.str_register.addr_reg_index] + cur_opcode.str_register.rel_address);
        break;
```

The code uses `addr_reg_index`, while the document says "5: Memory Region + Relative Address + Offset Register".

I guess it's more natural to use `ofs_reg_index` instead. But the code has exists for years. What do you think? @SciresM

SciresM, 2023-05-16:
So, code is the source of truth here, because we can't break cheats.

I guess the documentation should be updated to match.

DKingAlpha, 2023-05-16:
[Disassembler is here](https://nx.tot.im) if you care.

closed 2023-05-16 by DKingAlpha

SciresM, 2023-05-16:
Nice :)