Atmosphere-NX/Atmosphere #2134: Heat Issues In Emunand While Docked

Trilkin, 2023-05-20:
## Bug Report

### What's the issue you encountered?

While in emunand and docked, the heat of the system rises as if constantly under load. This will happen even if it's freshly booted into emunand with no suspended software and in sleep mode.

### How can the issue be reproduced?

- Boot into emunand while docked.
- Wait.

### Crash Report

N/A

### System Firmware Version

Happens in 16.0.1, 16.0.2 and 16.0.3.

### Environment?

- Booted through Hekate.
- Prerelease 1.5.4, but also happens in 1.5.3
- Tesla is the only module running, but happens even when it's removed
- Moonlight and Chiaki are the only homebrew I have installed
- Only Emunand

### Additional context?

- I have two USB devices connected through the dock: the ethernet adapter and the 8bitdo dongle.
- Potentially related if this ends up being a power-related issue, but Bluetooth reception is terrible in emunand while docked and only under those circumstances. If connected via wall wart or undocked entirely, it works fine at the same distances. USB through the dock is entirely unaffected as well. I get my full reported speeds through ethernet and I have no issues with the dongle.

CelsiusVi, 2023-05-27:
I concur.
Especially since I updated to 1.5.4 for Tears romfs mods, my switch has been overheating both in dock and while asleep outside of the dock (it feels particularly bad when it's in the carrying bag) and battery drains a lot faster while on sleep, as consecuence.
I deleted all sysmodules I'm aware of and it still feels like it is overworking itself, with no games running nor anything else going on.

System Firmware Version
16.0.3.

Environment?
Booted through Hekate.
Prerelease 1.5.4
Only Emunand

Additional context?
No USB devices conected, no more bluetooth devices connected. I once tried to charge it while in handheld mode and it caused the battery to drain faster even when it was receiving power.
Hekate Battery check looks pretty normal, 4200~mAh capacity, 400-500 power draw.

orio311, 2023-08-15:
@CelsiusVi Did you ever find a solution other than disabling DNS_MITM? I'm facing the same issue but can't afford to risk a ban

ELY3M, 2023-08-21:
My switch do get hot.   I run emunand on latest firmware and AMS.  I keep switch docked all of the time.  I run animal crossing treasure island and I have to leave it open all of the time.    I do take a break to let it cool down and charge it.

ELY3M, 2023-08-21:
I have DNS_MITM disabled. I will verify it again.

closed 2025-04-03 by SciresM