Atmosphere-NX/Atmosphere #2136: Support for usb wireless headset has broken in version 1.5.2

leafarilongamor, 2023-05-22:
## Bug Report

Support for usb wireless headset has broken in version 1.5.2

### What's the issue you encountered?

I have an old Sony playstation wireless headset (1st version) that had always worked with Nintendo Switch by connecting its dongle to the USB port in the dock. Since I updated to Atmosphere 1.5.2 this doesn't work anymore only on the Emunand. Works fine in the original Switch FW.

### How can the issue be reproduced?

Just have version 1.5.2 | 16.0.2 installed and try to use a Sony Playstation wireless headset (or possibly any wireless headset that uses 2.4ghz connection) with a USB 2.4ghz dongle on Nintendo Switch's dock. 
It won't work.

Then do the same on the original FW.
It works fine.

### System Firmware Version

16.0.2|1.5.2

SciresM, 2023-05-22:
Does this work in sysmmc? Emummc doesn't interact with anything that could have an impact here, to my knowledge.

ndeadly, 2023-05-22:
Given the timing of the issue, I would say it's highly likely @leafarilongamor uses Mission Control and hasn't updated to the latest version. The previous version (v0.9.1, released around the same time as ams 1.5.2) accidentally overrode the default USB interface filter put in place by Nintendo and prevented HOS modules from receiving USB events. This is fixed in v0.9.2.

leafarilongamor, 2023-05-23:
Hello All,
Thank you for checking this report. I will look for update the Mission Control in my Switch and check if it will solve the issue. I will let you know. I will also give the same direction to folks asking about the same issue in reddit.

leafarilongamor, 2023-05-23:
I have updated Atmosphere to 16.0.3|AMS 1.5.4|E and Mission Control to 0.9.2, however the issue persists. I will try to provide more information since in my original post I was on a hurry:

I have used since forever a Sony Platinum Wireless Stereo Headset (model CECHYA-0090). It was working on my V1 Switch Emummc until I updated to 16.0.2|AMS 1.5.2|E. It works normally when I boot into the Original FW/Sysmmc.

Proper behavior: When you connect the dongle to the USB port on the dock a small popup will appear at the switch UI indicating that the auto output is now directed to the USB. The dongle will start to blink indicating that is looking for the paired 2.4ghz headset. Turn on the headset and it will quickly blink as well for 2 seconds until both dongle and headset leds become static, indicating that is connected. Now the auto is working via headset.

Observed behavior: When you connect the dongle to the USB port on the dock a small popup will appear at the switch UI indicating that the auto output is now directed to the USB. The dongle will start to blink indicating that is looking for the paired 2.4ghz headset. Turn on the headset and both devices will blink forever and the headset will not reproduce any sound.

There's another user reporting the issue in reddit. I will not place the link here to avoid any harm to the project but you can easily find it by searching for: "USB headset problem using switch CFW"

Unfortunately I don't have another USB audio output device to test, but I believe the result will be the same, since it works Normally on the original FW. Before updating and starting to experience the issue, I was using version 14.1.2|AMS 15.0|S which also was supporting this use fine.

Thanks for any help you can provide. I would be interesting to try to reproduce the issue with a similar setup. I'm gonna touch base with the community and see if anyone can perform tests with similar setup.

SciresM, 2023-05-23:
If it works in OFW, to check: does it work if you completely remove mission control and any other custom sysmodules?

It it does, can you please post a screenshot of your atmosphere/contents to verify that there's no custom modules at all?

leafarilongamor, 2023-05-24:
I have uninstalled Mission Control using the procedure described at https://github.com/ndeadly/MissionControl. The issue persists.

Here is the screenshot and list of folders inside of atmosphere/contents in my SD card:

00FF0000A53BB665
00FF0000636C6BFF
054e4f4558454000
0000000000534C56
42000062616B6101
0100000000000F12
420000000007E51A
0100000000000352
420000000000000E
430000000000000B
690000000000000D
4200000000000000
4200000000000010

 [image]

Unfortunately I don't know which ones I can get rid of. If you see that I need to remove any of these, please let me know and I will remove them and try it again.

SciresM, 2023-05-24:
That is so many custom sysmodules.

Please do a test where you delete literally all of them, and there is nothing in the folder.

You can back them up somewhere else temporarily.

leafarilongamor, 2023-05-25:
I have wiped all  the contents folder and the issue persists.
[image]

By the issue tracker link you posted it is likely the same issue I'm facing. I'm gonna check how can I disable the sys-con and see if it works. I only use original controllers anyway. I let you know once I test it.

Again, thanks a lot for checking this out.

leafarilongamor, 2023-05-25:
I haven't figured out how to deactivate the sys-con. I removed the sys-con altogether from the config folder but the issue persists.
I tried to connect the USB dongle directly in the Switch's usb c port using a usb a to usb c adapter and it works fine. 🤷🏽‍♂️

leafarilongamor, 2023-05-25:
Alright I am now suspecting of a kind of hardware malfunction in the dock. I tested in another TV with another dock and it worked. Got back to my dock, started to mess a bit with the dongle in both usb ports until I finally heard the bip indicating connection and the led blue light got static. 

I still don't understand why it worked fine in OFW.... Maybe a coincidence that when I took the switch from the dock to reboot  it in OFW the usb contacted correctly. 🤷🏽‍♂️

For anyone facing the same issue, please double check all the involved devices with spare ones.

Thanks for the help and have a good one.

Masamune3210, 2023-05-25:
What I dont get is how running atmosphere would even possibly affect the headset from connecting to the dongle. From your description it sounds like it never does, since you said that they continually blink, meaning that they never stop searching for each other.

leafarilongamor, 2023-05-25:
@Masamune3210 , that was the initial symptom. When I noticed the issue, the second thing I tried to do was to boot in the OFW to check if it was working, and it did. 

Now I just kept moving the dongle in the dock's usb and it started to work. I will try to reproduce the issue again but I suspect of a hardware issue like bad contact in the usb along with a coincidence that when I tested the OFW the contact was correctly made between the dongle and the USB. It is though strange that the first thing I tried was to remove the dongle and reinsert it in both usb ports, to no avail. 🤷🏽‍♂️

I wonder if the other folks reporting the issue could have the same conditions and results.

leafarilongamor, 2023-05-25:
Ok. One of the USB ports (the one I've always used) simply don't work anymore (with CFW). The dongle keeps eternally blinking in search of the headset. The other port works but only after I press the dongle into it for a bit.

Last test (it doesn't make much sense but...): Booting in OFW and testing both usb ports.
Result: OMG it works with both usb in OFW. I don't know what to think.

Why in world the CFW would make one of the usb ports behave like it's defective? This reality is a lie, nothing makes sense. Lol.

Masamune3210, 2023-05-25:
Welcome to tech, it can have a mind of its own. It could be a number of things really, but honestly, they all pretty much point to that port just being slightly defective.

leafarilongamor, 2023-05-25:
Could it be that the CFW activates stuff that drags a little more power from the switch, interfering with the power that goes to usb ports? It is only logical explanation I can conceive for this.