Atmosphere-NX/Atmosphere #2141: Resident Evil Revelations (ID:0100643002136000) crashes after booting the game up.

x3DGamerx, 2023-05-26:
## Bug Report

Resident Evil Revelations' loading process doesn't work. When I start the game, I can see Nintendo's logo on the screen, as intended. However, a couple of seconds after that, the game completely crashes and the only thing I can read is: "The software was closed because an error occurred".

### What's the issue you encountered?

I purchased this game on the Nintendo eShop last year. It works like a charm on OFW. Let me point out the fact that this game has a special loading screen. Normally, you would expect to see Nintendo's logo after booting the game up. That happens as intended, but a whole new screen is thrown shortly after, inviting the player to start a minigame while the game keeps loading in the background. It's a retro-minigame, something you wouldn't expect to see.
Anyway, I wanted to install this title on emuMMC, too. So I did, but I had a nasty surprise: Atmosphère couldn't run the loading process and the game crashed, without giving further details.
I decided to test the error on Sysnand CFW. Probably not the best move I could do, but I tried anyway (I thought it was worth a try, since I legittimely bought the game). The error was there, ready to welcome me. I tried to dump the game (sysnand CFW) and install it again on emuMMC. Needless to say that didn't fix the issue.

### How can the issue be reproduced?

You can try it on your end. You can just install the title and run the game.

### System Firmware Version

16.0.3.

### Environment?

- What bootloader was Atmosphère launched by? Hekate.
- Atmosphère version: 1.5.3., official release.

masagrator, 2023-05-26:
Don't omit points of template. For example do you have any custom sysmodules installed.

x3DGamerx, 2023-05-26:
> Don't omit points of template. For example do you have any custom sysmodules installed.

No, I don't.

SciresM, 2023-05-26:
Please post a screenshot of the atmosphere/contents folder

SciresM, 2023-05-26:
> No, I don't.

This was a lie (though probably out of ignorance).

You have many custom sysmodules installed.

Does the issue go away if you remove your custom sysmodules? You can do this by temporarily renaming/deleting atmosphere/contents.

SciresM, 2023-05-26:
Is there a crash report in atmosphere/crash_reports?

x3DGamerx, 2023-05-26:
Yes, a couple of them. I wasn't sure which one I had to pick. If it's okay for you, I'll share all of them.
[01684963725_01aa7cb9f4128000.log](https://github.com/Atmosphere-NX/Atmosphere/files/11575983/01684963725_01aa7cb9f4128000.log)
[01684360454_01aa7cb9f4128000.log](https://github.com/Atmosphere-NX/Atmosphere/files/11575984/01684360454_01aa7cb9f4128000.log)
[01684360416_01aa7cb9f4128000.log](https://github.com/Atmosphere-NX/Atmosphere/files/11575985/01684360416_01aa7cb9f4128000.log)

SciresM, 2023-05-26:
Those aren't crash reports for resident evil revelations (id doesn't match). If you delete every file under atmosphere/erpt_reports, then launch the game, is a new report generated?

x3DGamerx, 2023-05-26:
Hm...after checking them properly, I think those logs were generated by JKSV. I tried to "access" game data files and the app crashed. 
I don't think a new report will generate, but let me try...

x3DGamerx, 2023-05-26:
Nope. No reports...

SciresM, 2023-05-26:
Can you delete the game, re-download from the eShop on sysmmc, and see if that helps?

x3DGamerx, 2023-05-26:
->No reports. 
Referring to your question instead, since this situation doesn't happen on OFW, how much am I risking? I never went online after what happened and expecially after that error I received on sysnand CFW...

x3DGamerx, 2023-05-26:
I don't think the game is going to work anyway...

SciresM, 2023-05-26:
Nintendo only bans for piracy and cheating online. So long as you are not misbehaving, you are fine. Note also that we do not support piracy; if you downloaded the game from the eshop on sysmmc initially you are obviously fine to redownload it from the eshop.

All testing should be performed on sysmmc, to be clear; you can reclone eventually.

Also, I do not give a shit what you think will work, I am asking you to test things because it will give knowledge useful for assessing the situation.

With high certainty, this will be some issue with your environment; it is overwhelmingly likely that someone else would have reported an issue with this game in the last six years if there was one fundamentally, so we are establishing a baseline.

x3DGamerx, 2023-05-26:
I get it. I don't understand much of it, so I apologize if I write dumb questions/thought of mine. I'd like to understand better, that's why. 
Anyway, when you talk about sysmmc -> OFW/sysnanc CFW?

SciresM, 2023-05-26:
I always mean cfw, so yes I mean sysnand cfw. I haven't booted ofw on any of my consoles in over four years, and really only care about what happens when atmosphere is running.

x3DGamerx, 2023-05-26:
> Can you delete the game, re-download from the eShop on sysmmc, and see if that helps?

I did as you required. I re-downloaded the game from the eShop on Sysnand CFW. The issue is still present.

cucholix, 2023-05-26:
I have the game in cartridge, same TID, no issues on my end ams 1.5.4 ofw 16.0.3
OP try updating the game by highlighting it, press +, and update, there is a hidden update for some reason.

x3DGamerx, 2023-05-26:
> I have the game in cartridge, same TID, no issues on my end, ams 1.5.4 ofw 16.0.3, and I have a bunch of custom sys-modules active.
> 
> OP try updating the game by highlighting it, press +, and update, there is a hidden update for some reason. The game BID on my system is 99E56E5136A1316B.

If you're talking about game version 1.0.1, I already had that update installed.

x3DGamerx, 2023-05-27:
I updated Atmosphére to 1.5.4. The issue is still present.
This time, though, won't ever boot up. As I try to launch it, an error is thrown: "Unable to start software. Return to the HOME menu and try again".

x3DGamerx, 2023-05-27:
I updated Atmosphére to 1.5.4. The issue is still present.

SciresM, 2023-06-05:
There's been no crash report posted for Resident Evil: Revelations, and another user has reported that the game launches fine for them on the latest update. Not really sure there's anything to do here.

rblxcode123, 2023-09-19:
I've had this same issue, tried everything. Did you ever find a fix to this?

ImanCol, 2023-11-17:
Intenta entrar a modo Recuperacion y realizar un restablecimiento sin borrar datos de usuario. Tendra que volver a descargar sus titulos pero corregira muchos fallos de certificados, que probablemente sea su problema.

Kris123454, 2025-03-04:
Same here. I checked the integrity and it checks out. Its sad how SciresM denies any possibility of it beeing atmosphere's fauly

SciresM, 2025-03-04:
@Kris123454 I don't...deny any possibility of it being atmosphere's fault in some way. If I did, I'd have closed this issue.

There's just nothing I can really do...if no crash reports are being generated, and it doesn't occur for @cucholix or myself (and I can confirm the game loads fine for me...)

cucholix, 2025-03-09:
Tested the game again, it boots normally, no issues.
19.0.1 | ams 1.8.0|S

SciresM, 2025-04-03:
I will, hesitantly, keep this issue open, but it's not replicable for myself and others.

Kris123454, 2025-04-13:
I FOUND THE SOLUTION!!! the problem was dvr patches. move folder atmosphere/exefs_patches/am to /atmosphere or /

SciresM, 2025-04-13:
I am going to close this, then.

closed 2025-04-13 by SciresM