Atmosphere-NX/Atmosphere #2144: Black screen with Loading romfs for Etrian Odyssey HD

FuyuArare, 2023-06-03:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

zemixe, 2023-06-03:
You ignored and deleted the issues template thus providing zero useful detail that would a assist in diagnosing an issue. Ryujinx is also completely irrelevant.

FuyuArare, 2023-06-03:
Oh, I overlooked it earlier, sorry i am not a native English speaker.

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
I used ryujinx to unpack game files and modified some of the textures for mod production.
When I load game with the romfs file, the game cannot run, I can just find black screen.
[ Did you make any changes related to Atmosphère itself? ]</br>
No, I use atmosphere-1.5.4-prerelease-3cb54e2b4, hekate_ctcaer_6.0.4_Nyx_1.5.3, sigpatch
[ If so, make sure to include details relating to what exactly you changed. ]</br>
None

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
The game(Etrian Odyssey HD, Etrian Odyssey II HD, Etrian Odyssey III HD) use the file 'catalog.json' for crc checks. I use [AddressablesTools](https://github.com/nesrak1/AddressablesTools/releases) to modify this file. Subsequently, I discovered that as long as the romfs directory contains this file(\contents\01007fc017fe2000\romfs\Data\StreamingAssets\aa\catalog.json), the game will enter a black screen after startup.

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>
Atmosphère Crash Report (v1.7):
Result:                          0x4A8 (2168-0002)

Process Info:
    Process Name:                SaltyNX
    Program ID:                  0000000000534c56
    Process ID:                  0000000000000081
    Process Flags:               00000137
    User Exception Address:      0000005d7bec4000
Exception Info:
    Type:                        Data Abort
    Address:                     0000000000000400
    Fault Address:               0000000000000000
Crashed Thread Info:
    Thread ID:                   0000000000000114
    Stack Region:                000000622f526000-000000622f52a000
    Registers:
        X[00]:                   0000000000000400
        X[01]:                   0000001be424fd40
        X[02]:                   0000000000000400
        X[03]:                   0000000000000400
        X[04]:                   0000001be4250140
        X[05]:                   0000000000000800
        X[06]:                   0000000000000000
        X[07]:                   0000000000000000
        X[08]:                   0000000000000000
        X[09]:                   0000000000000000
        X[10]:                   0000000000000000
        X[11]:                   0000000000000000
        X[12]:                   0000000000000000
        X[13]:                   0000000000000000
        X[14]:                   0000000000000000
        X[15]:                   0000000000000000
        X[16]:                   0000000000000000
        X[17]:                   0000000000000000
        X[18]:                   0000000000000001
        X[19]:                   0000001be424e988
        X[20]:                   0000000000000400
        X[21]:                   0000000000000400
        X[22]:                   000000000005cfb8
        X[23]:                   000000000005cfb8
        X[24]:                   0000001be423a470
        X[25]:                   000000000005d3b8
        X[26]:                   000000000005d3b8
        X[27]:                   0000001be424fd40
        X[28]:                   0000000000000400
        FP:                      000000622f5299f0
[... 128 lines cut ...]

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>
16.0.3|AMS 1.5.4|E & 16.0.3|AMS 1.5.4|S

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]
Sorry I cant understand what other content appears here.
My bootloader is from here (https://github.com/Yuanbanba/homebrew-NRO/releases )
I dont use homebrew software(I think)
I use EmuMMC.