Atmosphere-NX/Atmosphere #2145: Pre release 1.5.4 & tears of the kingdom mods

InkDragon, 2023-06-04:
## Bug Report

### What's the issue you encountered?
Crash upon launching Tears of the kingdom 1.1.2 on fw 16.0.3 with any modded content
[ Describe the issue in detail and what you were doing beforehand. ] fresh boot into cfw sysnad
[ Did you make any changes related to Atmosphère itself? ] no

### How can the issue be reproduced?
I'm using the physical version to load the game. Download any mod for version 1.1.2 and place in the modded files within "atmosphere\content\0100F2C0115B6000" on the SD card. boot into the cwf sysnand and attempt to boot the game. Produces a black screen with yellow text:

> Panic Occurred while running atmosphere
> Title ID: 010041544d530000
> Error: std: :abort (0xFFE)
> 
> Press POWER to continue.

please note no folder 010041544d530000 can be found on the SD card
### Crash Report
no crash report was generated

### System Firmware Version
16.0.3

### Environment?
- What bootloader? hekate 6.0.4
- Official release or unofficial build: Pre-release 1.5.4
- Do you have additional kips or sysmodules you're loading: Yes
- Homebrew software installed: on SD card but not directly on sysNAND. IE hbl, simple mod manager, JKSV, edizon, emuiibo, aio-switch updater
- EmuMMC or SysNAND: SysNAND

### Additional context?
- Additional info about your environment: previously used deepsea packs but switched to downloading most updates either manually via pc or aio.

masagrator, 2023-06-04:
> * Do you have additional kips or sysmodules you're loading: Yes

which ones

InkDragon, 2023-06-04:
kips i honestly dont know what all I have, but I know I have some. 
sysmodules installed are noexes, sys-clk, ldn,mitm, sys-patch(off), sys-con, 430000000000000B(sys-bot), sys-ftpd-10k(off), sys-ftpd-light(off), tesla, emuiibo, missionControl

SciresM, 2023-06-04:
You simply have too many sysmodules. There is not enough memory for all of those sysmodules and also zelda mods.

You can remove all of them, and zelda mods will work; you can then figure out what a minimal set you can have simultaneously with zelda mods is by trial and error. You can also reach out to sysmodule devs to see about reducing memory footprint, if any of them are too big on their own.

closed 2023-06-04 by SciresM