Atmosphere-NX/Atmosphere #2146: CFW not loading on new Switch

KamiSquadX, 2023-06-04:
## Bug Report

"Error Code: 2001-0123 (0xf601)
Program: 420000000007E51A
Firmware: 16.0.0 (Atmosphere 1.5.3-master-8e042f226)

An error has occurred."

### What's the issue you encountered?

CFW failed to start on both sysMMC and emuMMC, on a new Switch.

### How can the issue be reproduced?

Starting Hekate and Launching sysMMC or emuMMC

### Crash Report

"Error Code: 2001-0123 (0xf601)
Program: 420000000007E51A
Firmware: 16.0.0 (Atmosphere 1.5.3-master-8e042f226)

An error has occurred."

### System Firmware Version

Don't know, it says System update is ready.

### Environment?

[ [Atmosphère 1.5.3](https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.5.3) ]

- Do you have additional kips or sysmodules you're loading:
        How do I find out if I have any loading? (I do have Smash mods, and Themes, but this is a new Switch using my old SD Card with newly added emuMMC and sysMMC.
- Homebrew software installed: [ * ]
        Don't know, how do I find out if any are installed?
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]
            EmuMMC, partition-based. And SysMMC/SysNAND, both EmuMMC and SysMMC/SysNAND give an error when trying to load

### Additional context?

This is a new Switch, using my old Switch's SD Card which previously had it's own EmuMMC and SysMMC. I used the guide (https://nh-server.github.io/switch-guide/user_guide/emummc/launching_cfw/) to create new eMMC backups and install the updated files in the guide, as well as create a new EmuMMC and copy the new Nintendo folder into the EmuMMC folder.

However, I did not start from scratch, wiping the SD Card and creating a new partition.

Most of the files are still there, including the partition used by the old Switch's EmuMMC.

"Error Code: 2001-0123 (0xf601)
Program: 420000000007E51A
Firmware: 16.0.0 (Atmosphere 1.5.3-master-8e042f226)

An error has occurred."

masagrator, 2023-06-04:
> 420000000007E51A

delete this folder from atmosphere/contents

KamiSquadX, 2023-06-04:
Thank you. Will try it now.

Any idea why it was giving an error?

masagrator, 2023-06-04:
You had outdated Tesla Menu Installed

KamiSquadX, 2023-06-04:
Cool. Thanks. It worked.

closed 2023-06-04 by KamiSquadX