Atmosphere-NX/Atmosphere #2148: Redirect NTP to a public pool

apassiou, 2023-06-08:
## Feature Request

Currently with blocking nintendo servers via DNS or otherwise switch is unable to sync time. It would be nice if Atmosphere redirected (or did its own sync once a day or something) to a public ntp server such as pool.ntp.org

Otherwise the time has to be set manually and synced manually via NRO apps.

SciresM, 2023-06-08:
Can dns_mitm not be used for this purpose?

But also, I would point out that my expectation is most users of atmosphere are *not* blocking servers in DNS -- the median usecase should be sysmmc and going online.

apassiou, 2023-06-08:
> Can dns_mitm not be used for this purpose?
> 
> But also, I would point out that my expectation is most users of atmosphere are _not_ blocking servers in DNS -- the median usecase should be sysmmc and going online.

Possibly dns_mitm could do (never used it), but its not clear what NTP server switch uses (if this information was published then those of us redirecting nintendo server could simply redirect that one to pool.ntp.org).

With atmosphere is it trivial (or is it impossible) to simply overwrite nintendo's time sync server with pool.ntp.org?

SciresM, 2023-06-08:
I don't actually know how that works, to be honest. I'm kind of inclined to say that this should be achievable with dns_mitm, and if it's not you could probably just make a custom sysmodule that polls the ntp server of your choice and sets the network clock time. 

I'm not really sure this makes sense as an integrated-into-upstream-atmosphere feature.

apassiou, 2023-06-08:
ah ok, well, keep it in your mind in case you decide to do it :)

in meantime, can you surface the servers used by Nintendo? Or maybe give some pointers where to look?

apassiou, 2023-06-08:
Well it appears that the server is `aauth-%.ndas.srv.nintendo.net/v1/time` however, redirecting this to an NTP server does not seem to be working.

SciresM, 2023-06-08:
@apassiou are you taking into account the fact that "aauth-%" will have the % replaced by your environment identifier, which should be lp1?

For a retail console, the true URL there would be `aauth-lp1.ndas.srv.nintendo.net/v1/time`

apassiou, 2023-06-08:
Yes I am using `aauth-lp1.ndas.srv.nintendo.net/v1/time` in **mitm**

SciresM, 2023-06-08:
Well, that's a full url. Presumably you only want to redirect the base aauth hostname? You may also need to disable clcert verif (there's some patches for this, idr) but I'd expect your server to see the requests coming in.

apassiou, 2023-06-08:
yeah I tried base URL as well, not sure how to disable cert etc though

SciresM, 2023-06-16:
Going to close this, since it can be accomplished via extension sysmodule/etc.

closed 2023-06-16 by SciresM