Atmosphere-NX/Atmosphere #2149: Donkey Kong Country: Tropical Freeze crashes every time

moromarino, 2023-06-12:
This game crashes every time you try to get to the map screen
Examples: each time you finish a level, each time you load your savegame after the control method selection screen, or getting to the map after exitting a level). No cheats or mods involved…
If you want to reproduce this situation, you can just start a new game and try to exit mid-level to the map screen.
It crashes every time the screen is covered with clouds.

moromarino, 2023-06-12:
UPDATE!
It is fault of one of my sys modules. I renamed atmosphere/contents folder and everything works as usual.
I’m sorry if I wasted your time.
Thanks
Keep up the good work!

moromarino, 2023-06-12:
For your knowledge, it was SaltyNX’s fault

Masamune3210, 2023-06-13:
If you solved the issue, please close the issue report.

closed 2023-06-13 by moromarino