Atmosphere-NX/Atmosphere #2150: Unable to load payloads after using Atmosphere USB File Transfer feature

steveshi7, 2023-06-12:
## Bug Report

### What's the issue you encountered?
I attempted to use Atmosphere 1.5.3's USB File Transfer feature, but the Switch was not detecting that it was connected to my computer (nor mounting).

I then attempted to restart the Switch to use Hekate's USB feature, which has worked in the past. However, attempting to load hekate using fusee-launcher resulted in a consistent timeout error:
```
Identified a macOS system; setting up the appropriate backend.
Found a Tegra with Device ID: [device id]

Setting ourselves up to smash the stack...
Uploading payload...
Traceback (most recent call last):
  File "/Users/steve.shi/co/fusee-launcher/fusee-launcher.py", line 601, in <module>
    device_id = switch.read_device_id()
  File "/Users/steve.shi/co/fusee-launcher/fusee-launcher.py", line 543, in read_device_id
    return self.read(16)
  File "/Users/steve.shi/co/fusee-launcher/fusee-launcher.py", line 500, in read
    return self.backend.read(length)
  File "/Users/steve.shi/co/fusee-launcher/fusee-launcher.py", line 118, in read
    return bytes(self.dev.read(0x81, length, 1000))
  ...
  File "/opt/homebrew/lib/python3.10/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 60] Operation timed out
```

Since my Switch is in AutoRCM mode, this has effectively bricked my Switch as I am unable to turn it on without loading a payload. This has happened before in the past, but I was able to recover by holding the power button for 15 seconds. For some reason, this is no longer working.

I believe this may have been caused by the USB File Transfer feature itself because the issue is specifically correlated with attempting to use the feature, ie I was able to restart my Switch fine multiple times after the first time this happened, and this did not occur again until attempting to use the feature again.

### How can the issue be reproduced?

1. Attempt to use USB File Transfer feature, which does not mount to laptop
2. Restart device
### Crash Report

N/A

### System Firmware Version
16.0.3

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee, but also tried FuseeWebLauncher, CrystalRCM
- Official release or unofficial build: 
  - 1.5.3
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Hekate 6.0.3
- EmuMMC or SysNAND:
  - SysNAND

### Additional context?
- AutoRCM is on

steveshi7, 2023-06-12:
> Atmosphere's file transfer feature has nothing to do with it. Its either a broken cable, a broken USB port on the console, or some kind of software change on your PC (doubtful considering standard protocols like MTP didn't work)

I agree that it feels pretty unlikely; I guess I'm confused because it feels like there are no other confounding variables?

Timeline:
- This happened two weeks ago after unsuccessfully using file transfer feature. Fixed by hard shutting down switch.
- In the following two weeks, rebooted multiple times. Used Hekate UMS multiple times during this time. 
- Last night, tried again to use file transfer feature, which was again unsuccessful. Unable to inject payload this time.

All other variables are definitely the same: cable, port, PC. Have also tried other ports and cables unsuccessfully.

Appreciate any other ideas that you may have here.

closed 2023-06-12 by steveshi7

Masamune3210, 2023-06-12:
It honestly sounds like either the cable or one of the devices/ports is the issue here. All the issues line up with that, the fact that transfers wouldn't work, the fact that payloads cant be sent correctly, which also involves USB transfer. Best thing to do would be to break it up into chunks to test, try a different host with a different cable to make sure the payload can be sent to the switch correctly, then if that works, try it with the different host and the same cable, then if that works, try it with the old host. If that doesn't work, you know either the software or the hardware of the host is the issue.