Atmosphere-NX/Atmosphere #2151: Error Code 2001-0123 (0xf601) - Atmosphere 1.5.3 master crashes on boot after upgrading to firmware 16.0.3

lurondev, 2023-06-14:
## Bug Report

See the attached screenshot 
[image]

### What's the issue you encountered?

I started at 13.0.0 | AMS 1.3.2 | E and tried to upgrade to the latest using:
- Upgraded hekate_ctcaer_6.0.5_Nyx_1.5.4_v2
- atmosphere-1.5.3-master-85c23b578+hbl-2.4.3+hbmenu-3.5.1
- Sigpatches.zip

Everything worked up to this point. I tested my games, no problems. The system was at 13.0.0|AMS 1.5.3|E

Went through update to 16.0.3 using 
Firmware 16.0.3 (Rebootless Update)
Update was actually successful. Once the system rebooted the error was presented

Looks identical to the closed issue:
https://github.com/Atmosphere-NX/Atmosphere/issues/1437

lurondev, 2023-06-14:
Also, pre-update I backed up my entire SD card folders except: emuMMC / Nintendo / switch. 
Can I replace the current SD card contents with the pre-update backup and go back to initial state of 13.0.0 | AMS 1.3.2 | E ?

lurondev, 2023-06-14:
[01686701803_420000000007e51a.log](https://github.com/Atmosphere-NX/Atmosphere/files/11740864/01686701803_420000000007e51a.log)

Attaching a fatal report - could not find others

mhussle, 2023-06-14:
Go check into your SD Card into : atmosphere\fatal_reports

And check the last report in "Program ID : XXXXXXXXXXXXXXXX"

Mine was "420000000007e51a"

Then, go check into : "atmosphere\contents"
And delete the folder that have the same name has the Program ID Showed to you and normally it's all done (Stolen from another thread but I did this as well and it worked)

closed 2023-06-14 by lurondev

Scavy74, 2023-08-07:
Same error for me, could be solved with this solution, thanks :)

NikoHuerta, 2023-08-16:
same error here, the comment up there has the solution !

ivan8348, 2023-08-18:
> Go check into your SD Card into : atmosphere\fatal_reports
>
> And check the last report in "Program ID : XXXXXXXXXXXXXXXX"
>
> Mine was "420000000007e51a"
>
> Then, go check into : "atmosphere\contents" And [...]

Thank you mhussle! It worked!!!!!

mhussle, 2023-08-18:
> Thank you mhussle! It worked!!!!!

I’m so glad it worked! I was very frustrated myself. Cheers

Makochang0705, 2023-08-28:
It did not work on my switch....
Please help!!! And thank you
[image]

BeboBabaYellow, 2024-01-23:
@Makochang0705 you need to wait for the image to upload before sending, it just says "uploading IMG_XXXXXXXX_XXXXXX.jpg..." for me (X = Number)

bradleytechman, 2024-04-03:
> Go check into your SD Card into : atmosphere\fatal_reports
>
> And check the last report in "Program ID : XXXXXXXXXXXXXXXX"
>
> Mine was "420000000007e51a"
>
> Then, go check into : "atmosphere\contents" And [...]

Thank You!!!!

gato911-source, 2024-12-07:
literally the same error, thank you mhussle!!!!