Atmosphere-NX/Atmosphere #2152: Orange screen or error code while playing games or put the switch in sleep mode

glicheur123, 2023-06-14:
## Bug Report

### What's the issue you encountered?

When I play a game in coop mode, the console randomly freeze and shows an orange screen or the code 2162-0002 associated to the  program 01000000000100C (log included) </br>

The orange screen occur when I put the console in sleep mode. </br>

however, if I play the same game in single mode (like mario bros u), the switch is working fine. 

 
### How can the issue be reproduced?

After we put the switch on the dock, the console randomly crash or freeze when we use the joycons for selecting a character.  </br>

Note: This issue can be reproduce on various multiple game (so far, for 3 games I tried, 2 of them crashed).

### Crash Report

[For crash reports](https://gist.github.com/glicheur123/a77bfc55b0143ed66eb2cb35b6b0cc7a) </br>
[For fatal reports](https://gist.github.com/glicheur123/8ccbb361743440aceed2b78ffaa925f7)

### System Firmware Version

16.0.3 AMS 1.5.3</br>

### Environment

- bootloader : Hekate
- Official release v6.0.4
- No additional sysmodules installed
- Homebrew software installed: daybreak, awoo Installer, Oi updater
- Loaded by EmuMMC with a partition-based

glicheur123, 2023-06-16:
> Orange screen = hardware issue. Try in OFW

Already tried and the orange screen occured. Which component can produce this issue? I think the sd card or maybe the chip installed on the switch. But that is not explain why atmosphere randomly crash when I started a game on the dock. In the OFW when the console crashed, the orange screen shows up but the CFW shows an error 2162 or a orange screen. Why?

SciresM, 2023-06-16:
Yeah, orange screen indicates a hardware problem. If that's occurring, not really anything to be done from Atmosphere's side.

SciresM, 2023-06-16:
The issue tracker is for atmosphere bugs, not for issue support.

closed 2023-06-16 by SciresM