Atmosphere-NX/Atmosphere #2154: Tears of the Kingdom infinite loading screen with mods installed

JacobVEyck, 2023-06-15:
## Bug Report

### What's the issue you encountered?

When mods are removed everything runs fine, when they are installed the game never crashes, it just hangs on the loading screen indefinitely, I can interact and press a to cycle through the tips and such, so the game is still running. 

No changes made to atmosphere, just updated today for the purpose of modding TotK.

### How can the issue be reproduced?

This happens every time I try to load any save file

### Crash Report

N/A
### System Firmware Version

16.0.2

### Environment?

fusee launched through hekate

### Additional context?

I uninstalled all overlays and themes, running very bare bones.

SciresM, 2023-06-15:
That sounds like an issue with your mod, not with atmosphere. Talk to the mod creator.

fwiw, once it's in-game, atmosphere's part is done (and the mod working is up to the mod creator).

closed 2023-06-15 by SciresM

JacobVEyck, 2023-06-15:
I tried it with multiple mods while having only one installed at a time and the same thing happened every time.

SciresM, 2023-06-15:
I really can't help you with this one, except to, I dunno, suggest a different SD and check that your archive bits are clear.

Once the game has loaded at all, atmosphere won't make any difference, and totk mods do work at all (per reports).