Atmosphere-NX/Atmosphere #2155: Error code 2002-4373

braybray124, 2023-06-15:
## Bug Report
Program 010000000024
When I boot up my switch with us chipped withw the hwfly rp2040 an when it is going to boot in the emmnand I get the error code after the Nintendo logo an I put in on the sd card with the NEXT aio pack an it an it is working fine ani can not use it anymore so can you pls help fix this

Masamune3210, 2023-06-15:
Please go back and actually fill out the issue template.....

Cam396, 2023-06-16:
if you want a guaranteed working homebrew download the latest release of Atmosphere (https://github.com/Atmosphere-NX/Atmosphere/releases) hekate (https://github.com/CTCaer/hekate/releases/tag/v6.0.5) and grab some packages on sdsetup.com

SciresM, 2023-06-16:
This didn't use the issue templates.

closed 2023-06-16 by SciresM