Atmosphere-NX/Atmosphere #2157: Black screen after booting to Atmosphere

Switchbruh, 2023-06-17:
Hey guys, been using Atmosphere for years at this point and never encountered any issues

Just updated to 1.5.4 and booted into Atmosphere, the Atmosphere logo flashes up and its like it's booted into the switch CFW but its a black screen, the only thing I can do is hold power and reboot to hekate. I've tried booting fusee directly from Tegra and it's the same

I've never had any issues before so please be gentle with my lack of knowing what other info might be needed to help

thanks

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ I was updating Atmosphere because MineCraft Legends wouldn't work (I kept getting the error 'the software was closed because an error occured') The game didnt work before I updated ]</br>
[ updated the atmosphere files]</br>
[ N/A ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ I'm not sure]</br>

### Crash Report

[ there are no crash reports regarding the black screen, there are crash reports for minecraft, but this isn't related to this issue I dont think. Atmosphere was loading and my other games worked . ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:

  - [ Official release version x.x.x (or) unofficial build ]  official v6.0.5 hekate

- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: [ no ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ] partitioned

### Additional context?

- Additional info about your environment:
- [i've tried downloading different versions down to 1.4.1 and replaced the files to see if this changes anything but it doesn't. still get a black screen]

Switchbruh, 2023-06-17:
Stock MMC still works also

Masamune3210, 2023-06-17:
If your SysNAND works fine but your EmuNAND doesn't work anymore, try backing up your current EmuNAND to somewhere safe and recreate your EmuNAND to make sure it isn't just a corruption issue or anything like that with the EmuNAND

Wamy-Dev, 2023-06-17:
same issue, was working right before updating to 1.5.4

CarlCross1, 2023-06-21:
Reset factory and new emunand

Wamy-Dev, 2023-06-26:
> Do a  factory reset and a new emunand 

ended up having to do this unfortunately

mikecheng2626, 2023-08-05:
> > Do a  factory reset and a new emunand
> 
> ended up having to do this unfortunately

Have you sorted out what the problem is?
I just followed re-entry guide for a clean emunand install. I have same issue. Started as hekate IPL loading bar --> atmosphere logo maybe like 1 sec blimp --> everything goes black --> had to power off switch and inject payload via Tegra again

All latest releases

duskshadowbrony, 2023-09-14:
I fixed the issue by just getting a brand new SD card. It seems that my old one had corrupt files that Windows could not delete properly, and that were not caught in the scan and repair.

Fresh SD card worked perfectly.

boubou4360, 2023-09-18:
same issue tried everything mentions above, everything up to date idk whats the problem tbh . tried to boot trough hekate with the fusee.bin(taking the same updated fusee.nin everywhere).My root folder looks like this. At this point might be my sd card. Nobody got an idea or a suggestions before wiping ?
[image]

duskshadowbrony, 2023-09-18:
> same issue tried everything mentions above, everything up to date idk whats the problem tbh . tried to boot trough hekate with the fusee.bin(taking the same updated fusee.nin everywhere).My root [...]

Your atmosphere folder hasn't been updated since 2020. Have you downloaded the newest version?

boubou4360, 2023-09-18:
Yes I did it just keep de modified date to 2020 idk why tho but I did it its weird

Masamune3210, 2023-09-19:
Move your existing SD card contents to a different place, set the card up again like you would if you were just setting up atmosphere, and see if it happens again.

PreMorph22, 2023-10-28:
No new install. Just fixed it by using another fusee.bin as payload to inject: https://github.com/Atmosphere-NX/Atmosphere/issues/2192

DereksDntRun, 2023-10-30:
I'm the same.
I was on FW16.0.3/1.5.5 on EmuMMC partition (on FAT32 SD card) on unpatched HAC-001.
I tried updating to 16.1.0 as I would normally:
- dl'd 16.1.0 and extracted to fw folder on sd
- update Hekate to 6.0.7 and that seemed good
- updated Atmosphere to 1.6.2 with latest sigpatches, put the card into the switch and booted into CFW all ok.
- R+game to launch non-applet mode -> Daybreak and installed 16.1.0 (accidently chose FAT32+exFAT - but I don't think that should matter).
- it said it installed fine, rebooted and it flashes Atmosphere logo for 1/2 second and only blank screen.

much swearing, tried 1.6.2's fusee and lots of different hekate_ipl.ini mentioned online, and always black screen.  Tried removing/renaming the Atmosphere/contents folder (as some have suggested), but still black screen.

- Tried OFW and that works, so it's not bricked the Switch.
- Went back to Hakate and restored EmuMMC back to FW16.0.3 and working fine.
- re-installed Atmosphere 1.6.2, this time from a AIO package - booted and worked fine.
- re-dl'd FW16.1.0 from a new location.
- non-applet Daybreak again and installed with FAT32 only option
- still the same - flash of Atmosphere logo and black screen.

I feel like I've tried everything apart from a clean format of the SD card.  But, I'd be restoring from the same FW16.0.3 backup, so I think the partition would still be in the same state?

I'm 1/2 wanting to try FW17.0.0 but with the comments about bricking loops/issues then I'm wondering if what's causing 16.1.0 to be issue will still occur on 17.0.0?

Is something dodgy?  or what more can I try?  Is 16.1.0 checking for fuses that aren't burnt via Daybreak, or something silly like that?

DereksDntRun, 2023-10-30:
^ as a follow-up from my post above, I finally got it working into 16.1.0.

I'm guessing there's something from an older Atmosphere install or an addon (not in contents folder) that's caused it.

to fix I effectively cleaned the SD card as much as I felt I could without a full format:

instead of just deleting, for safety I moved the files to my PC, but I removed everything apart from:
\emuMMC \backup \firmware and \Nintendo
(so, for my sd card, I removed \atmosphere \bootloader \config \retroarch \switch and contents of root).

- added a brand new clean install of Atmosphere AIO from https://sigmapatches.coomer.party/ (1.6.2)
- added Hekate (6.0.7)
- turned on Switch and injected hekate - set that up and fusee payload into a my emuMMC restore of 16.0.3
- non-applet Daybreak and installed 16.1.0, Fat32, preserve settings and shutdown.
- now seems to boot without locking on a black screen. 👍

SciresM, 2023-10-30:
Deleted commend containing warez stuff. Also, please do not coopt other people's issues.

KALRONG, 2023-10-31:
Same behaviour after updating atmosphere, hekate and sigpatches but NOT firmware to all latest versions; solution in my csae was rolling back the files in the atmosphere folder to the previous 1.5.5 ones I had, now emucc working again.

Stephen00c, 2023-12-23:
so what did you do to fix it bc im having this problem

boubou4360, 2023-12-23:
what worked for me its updating atmoshphere to another version and it worked, maybe the atmsh version i had wasnt compatible.

bmmiller, 2024-01-12:
> Same behaviour after updating atmosphere, hekate and sigpatches but NOT firmware to all latest versions; solution in my csae was rolling back the files in the atmosphere folder to the previous 1.5.5 ones I had, now emucc working again.
> 
> UPDATE: Took the chance and updated to 17.0 while on atm 1.5.5 and then update to 1.6.2 and the black screen no longer shows.

Thanks for including this reply.  I was having similar issues going from 1.5.4/16.0.3 -> 1.6.2 (before upgrading to 17.0.1) and just getting a black screen after the atmosphere logo.  On a prayer, rolled back to 1.5.4 and booted right up so came looking for answers.  Found your post, upgraded to 17.0.1 (didn't boot of course) and then manually upgraded atmosphere to 1.6.2 and boom, all good!

yeet-dot-org, 2024-01-13:
> > Same behaviour after updating atmosphere, hekate and sigpatches but NOT firmware to all latest versions; solution in my csae was rolling back the files in the atmosphere folder to the previous 1.5.5 ones I had, now emucc working again.
> > UPDATE: Took the chance and updated to 17.0 while on atm 1.5.5 and then update to 1.6.2 and the black screen no longer shows.
> 
> Thanks for including this reply. I was having similar issues going from 1.5.4/16.0.3 -> 1.6.2 (before upgrading to 17.0.1) and just getting a black screen after the atmosphere logo. On a prayer, rolled back to 1.5.4 and booted right up so came looking for answers. Found your post, upgraded to 17.0.1 (didn't boot of course) and then manually upgraded atmosphere to 1.6.2 and boom, all good!
> 
> (I had already upgraded sigpatches to the version that supported 1.6.2 when I downgraded to 1.5.4. Research told me they were backwards compatible so I did not search out what I was using for 1.5.4 and did not touch them again during this process)

i had a panic error show up when i booted to atmosphere, and after looking online i applied a fix by pressing "Fix Archive Bit" button in hekate. now i have a the same problem you had where after launching sysmmc cfw, the atmosphere would show for a few seconds and a black screen would appear. (weirdly, my PC said "setting up 'Nintendo Switch'/ 'Nintendo Switch' is ready to go" after the switch went to a black screen). I'll try downgrading Atmosphere to 1.5.4 never had a previous version of atmosphere as I was following a video guide on how to restore HOS using a donor nand.

baalbeker, 2024-01-15:
Had the same problem.
To avoid all headaches follow this:

Update Atmosphere through AIOupdater and after that update the firmware through Daybreak.This way there is no problem with black screen.

After that i got the Error 2123-0011 that is triggering when the wifi is on.
You can fix it by doing the Exosphere and DNS MITM settings from the Rentry guide.

After these fixes the switch is again top notch.

yeet-dot-org, 2024-01-16:
> Had the same problem on switch lite. To avoid black screen follow this:
> 
> Update Atmosphere through AIOupdater and after that update the firmware through Daybreak.This way there is no problem with black screen.
> 
> After that i got the Error 2123-0011 that is triggering when the wifi is on. (Nintendo did some NEXT LEVEL PIRATE PREVENTION TACTICS) You can simply fix it by doing the Exosphere and DNS MITM settings from the Rentry guide.
> 
> After these fixes the switch is again top notch.
> 
> [ADDITIONAL INFO] On update 17.0.1,the black screen on boot appears when you update atmosphere manually.Thus you need to update through AIOupdater.
> 
> To avoid being in bad situation after atmosphere update,first backup atmosphere,bootloader and switch folder on your pc in case you get black screen.This way you can pop out your sd card and overwrite the files and have your system working again with the previous atmosphere version.
> 
> "Im here to kick errors and chew ass.And im all OUTTA ERRORS"
I have actually fixed the issue by just redoing the YouTube guide but instead using firmware 16.1.0 that the YouTuber suggested to me. Thanks for the info, never knew there was such a thing as an AIO updater, interesting. Now my problem is that my switch is practically banned as i found-out/should've-realized it can't connect to the eshop or anything online as I used a donor prodinfo and nand as I was too incompetent to realize I should've backed up the nand before wiping it for lineage os android (long story). I don't really know what to use my practically forever banned switch for, maybe Linux, dunno really.

bmmiller, 2024-01-16:
> Had the same problem on switch lite. To avoid black screen follow this:
>
> Update Atmosphere through AIOupdater and after that update the firmware through Daybreak.This way there is no problem with [...]

I don't really agree with this methodology because it did not echo my experience.  Just FYI for others reading.

I initially upgraded atmosphere from 1.5.4 -> 1.6.2 using AIOUpdater and updated sigpatches manually.  Rebooted, expecting v1.6.2 atmopshere to still work with v16.0.3 of firmware.  This did not work.  This resulted in the black screen.

What fixed it was _manually_ downgrading back to 1.5.4.  Upgrading firmware to v17.0.1 with Daybreak and then manually upgrading to 1.6.2.  Manual upgrade to 1.6.2 at this point is required because CFW won't boot of course with new firmware and old atmosphere.

At no point was AIO Updater an actual solution for this issue.  It wasn't the problem.  But it wasn't the solution either.  If you update to v1.6.2 in AIO Updater (updating sigpatches however you want) and then reboot, in my experience, you will be dead in the water.  If AIO Updater allowed you to update to 1.6.2 and then you hopped over and upgraded firmware to 17.0.1 **without rebooting** until firmware was updated, that'd probably work, but AIO Updater forces a reboot after upgrading atmosphere in this scenario.

_Usually_ updating atmosphere + sigpatches ahead of firmware and rebooting is fine.  Something was different this time.  I did want to point out that in my experience, on my particular version path, there is no way to upgrading atmosphere with AIO Updater and then use Daybreak successfully.  It was not possible.

yeet-dot-org, 2024-01-16:
> > Had the same problem on switch lite. To avoid black screen follow this:
> > Update Atmosphere through AIOupdater and after that update the firmware through Daybreak.This way there is no problem with black screen.
> > After that i got the Error 2123-0011 that is triggering when the wifi is on. (Nintendo did some NEXT LEVEL PIRATE PREVENTION TACTICS) You can simply fix it by doing the Exosphere and DNS MITM settings from the Rentry guide.
> > After these fixes the switch is again top notch.
> > [ADDITIONAL INFO] On update 17.0.1,the black screen on boot appears when you update atmosphere manually.Thus you need to update through AIOupdater.
> > To avoid being in bad situation after atmosphere update,first backup atmosphere,bootloader and switch folder on your pc in case you get black screen.This way you can pop out your sd card and overwrite the files and have your system working again with the previous atmosphere version.
> > "Im here to kick errors and chew ass.And im all OUTTA ERRORS"
> 
> I don't really agree with this methodology because it did not echo my experience. Just FYI for others reading.
> 
> I initially upgraded atmosphere from 1.5.4 -> 1.6.2 using AIOUpdater and updated sigpatches manually. Rebooted, expecting v1.6.2 atmopshere to still work with v16.0.3 of firmware. This did not work. This resulted in the black screen.
> 
> What fixed it was _manually_ downgrading back to 1.5.4. Upgrading firmware to v17.0.1 with Daybreak and then manually upgrading to 1.6.2. Manual upgrade to 1.6.2 at this point is required because CFW won't boot of course with new firmware and old atmosphere.
> 
> At no point was AIO Updater an actual solution for this issue. It wasn't the problem. But it wasn't the solution either. If you update to v1.6.2 in AIO Updater (updating sigpatches however you want) and then reboot, in my experience, you will be dead in the water. If AIO Updater allowed you to update to 1.6.2 and then you hopped over and upgraded firmware to 17.0.1 **without rebooting** until firmware was updated, that'd probably work, but AIO Updater forces a reboot after upgrading atmosphere in this scenario.
> 
> _Usually_ updating atmosphere + sigpatches ahead of firmware and rebooting is fine. Something was different this time. I did want to point out that in my experience, on my particular version path, there is no way to upgrade atmosphere with AIO Updater and then use Daybreak successfully. It was not possible.

Man i wish github has some kind of flowchart feature. that could realy make it easier to comprehend the workaround that's needed to fix the black screen

Konngitas, 2024-02-24:
I am getting the same issue after updating to 1.6.2. On firmware 16.0.1. Happens with both sysMMC and emuMMC. Only thing that works is downgrading back to AMS 1.5.5.

Is AMS 1.6.2 broken?

bmmiller, 2024-02-24:
> I am getting the same issue after updating to 1.6.2. On firmware 16.0.1. Happens with both sysMMC and emuMMC. Only thing that works is downgrading back to AMS 1.5.5.
> 
> Is AMS 1.6.2 broken on firmware 16.0.1? I would rather not risk updating my firmware to 17.0.1 and be stuck without working CFW.

No idea but all I can say is my process worked.  I know it's a risk but it's not like you are risking bricking it or something.

Konngitas, 2024-02-24:
Figured my issue out. I forgot to update fusee haha. Nevermind me!

closed 2025-04-03 by SciresM

BedroVibbus, 2025-08-14:
Now i'm suffering this even with fusee updated, wasn't happening on the previous firmware (curent is 20.3.0, and it is already supported)