Atmosphere-NX/Atmosphere #2164: Black screen after system fan translation

tomxhachijuroku, 2023-06-27:
Hello, I translated Firmware 16.0.2 into polish, but as soon as I put all of the folders in the contents folder together, I got a black screen after the SWITCH logo, after some troubleshooting, I found out that Nintendo must have make some protection to unable a full fan translation, because when I put 0100000000001000 and 010000000000100C folders together with the rest of the translated folders the console won't go pass the SWITCH logo, even if they are completely empty (no romsfs, fsmitm.flag and romfs_metadata.bin), but if I delete or rename one of those folders the console boots normally, so for now I chose to use 0100000000001000 because it has more files inside like Home menu, Options etc. and 010000000000100C only has the overlay menus like Power Options, Battery Charge etc.

But It would be nice if the whole translation worked, so does any body know what is happening with those two folders, do they need a exefs .ips sigpatches, like the Lock Screen need from time to time in SwitchThemesNX, if so can you, fix that please, or tell me what I'm doing wrong?

Masamune3210, 2023-06-27:
I promise you, Nintendo did not put any special sauce in Horizon just to sniff out a random fan translation lol.
You didn't use the issue template.
Also, this place is not for support, only issues with Atmosphere itself; please go elsewhere to find support for your mod

tomxhachijuroku, 2023-06-28:
OK, so its not Nintendo's doing, that's strange my translations worked till 15.0.1, maybe it is the sigpatches, thank you for clearing that for me, like I said it's not a big deal I'm not trying to release my translation to the wild, I just want to learn new things.
I'll ask the creators of SwitchThemesNX, how to do the sigpatches, maybe that will help, Thank you again and have a good life.

PS. You don't have to be so rude I'm just asking for help, so chill.

Masamune3210, 2023-06-28:
I will be less rude when people become more respectful and literate. The template exists for a reason, this is not for general support, and piracy is explicitly banned from having support.....which is stated in the template you didn't read.

tomxhachijuroku, 2023-06-28:
OK, I'm sorry I forgot to mention that I'm not trying to release my translation to anybody, I'm just doing it form my self to learn something, just like I do with my ongoing game translations, that I also only doing for my self, because I don't want Nintendo to sue me, so thanks again and, bye.

JulyIghor, 2023-09-11:
> Also, this place is not for support, only issues with Atmosphere itself; please go elsewhere to find support for your mod

It is possibly related to Atmosphere anyway, if you just extract current firmware's 0100000000001000 and 010000000000100C  folders to atmosphere/contents folder, it won't boot like the post author said. Expected that it should not affect Horizon OS, because no files changed, but it is. Can you confirm that it isn't Atmosphere bug?

boni89, 2023-11-19:
salve ho aggiornato la Nintendo alla versione 17, ma le versioni di Atmosphere 16 16.1 e 16.2 dopo la partenza resta nera, ho letto che gia lo sapere ma è possibile fare qualcosa per ora? o devo solo sperare nella versione nuova?

SciresM, 2025-04-03:
Yeah, the 100C thing is a known problem that I have no idea how to fix, sorry.

closed 2025-04-03 by SciresM