Atmosphere-NX/Atmosphere #2165: error 2002-4373 (0x222a02)

xagus193, 2023-07-01:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br> program:010000000000000C
[ - software used to make "backups", ]</br>14.1.2
[ - software explicitly distributed for piracy, etc ]</br>atmosphere 1.5.3-master-8e042f226)
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered? 
crashes just start

[ Describe the issue in detail and what you were doing beforehand. ]</br>It is a crash as soon as it starts, before it happened I was playing Zelda Tears of the Kingdom
[ Did you make any changes related to Atmosphère itself? ]</br>no
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
turn on the console
[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. []</br>]https://gist.github.com/xagus193/a900174ae03d290c6143b1f81a049121
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

tommydurkan, 2023-11-20:
Just encountered same error, a 12-month-old switch, software V17.0 , froze while watching YouTube App and now gets this error at power on. No combination of holding buttons seems to resolve, local tech repair shop also failed to fix it.
Any ideas welcome.

Tthat1guy, 2024-07-01:
encountered this error running atmosphere version 17.0 on a 18.0 switch that has been hardware modded. I clicked a game installed with tinfoil (minecraft dungeons) and it loaded and then crashed saying something like "cannot load application" (I don't remember exactly what it said) then no games will load immediately giving the same error. then randomly the switch crashed giving the error code 2002-4373 (0x222a002) it could not be fixed by power cycling the switch or updating the atmosphere install to 17.1

JosueChoso, 2024-08-06:
Any solution? I have the same problem and I can't solve it. I could try to reinstall atmosphere but I don't wanna lose my saved games...
[image]

hunt3r0one, 2024-08-13:
Same error with the same output as @JosueChoso

BriseII, 2024-08-14:
@JosueChoso or anyone did you guys find a solution. I just installed a modchip on my switch but rcm wasn't on so i loaded payload thought Takate and i keep getting this error(2002-4373 (0x222a02)). Do i need to buy a rcm and load it through pc or is it something else.

JosueChoso, 2024-08-14:
@BriseII In the end, I had to back up the save data from the SD card and reinstall Atmosphere. To be safe, I kept the old SD card as-is and bought a new one that’s better quality.

BriseII, 2024-08-14:
so its an SD card problem? i've tried 3 different ones and installed the last 3 verisons of atmosphere, either i'm really unlucky or its a different problem. thanks for the answer.

Tthat1guy, 2024-08-14:
completely reformatting the sd card did not fix this problem for me. I'm going to try putting the nand back into its original place to see if the console boots.

hunt3r0one, 2024-08-14:
I doubt it's a problem with the SD card. Mine is brand new but still get the same error.

hunt3r0one, 2024-08-14:
Fixed by making a new emuMMC.

BriseII, 2024-08-15:
@hunt3r0one if you watched a video on it do you think I can get the link also how big is your sd because I’m using a 32gig which I don’t think is enough

hunt3r0one, 2024-08-17:
> @hunt3r0one if you watched a video on it do you think I can get the link also how big is your sd because I’m using a 32gig which I don’t think is enough

I didn't watch any video I just reformatted the SD card and created a new emuMMC partition with Hekate. Mine is 128gb, I would recommend at least that

closed 2025-04-03 by SciresM