Atmosphere-NX/Atmosphere #2166: RCMLoader error 16.0.3 atmosphere 1.5.4

GodIsSoGreat, 2023-07-08:
## Bug Report

### What's the issue you encountered?

Using RCMLoader and jig and renaming fusee.bin to payload.bin I get an error. 

This error:

"A fatal error occured when running Fusee.
Incorrect fuse version! (program=oxb46dcf71, mtc=0x4aa02b53)

Press POWER to reboot."

If I use tegraGUI to inject payload everything works fine.

### How can the issue be reproduced?
Put new fusee.bin on RCMLoader. slot in jig to right joycon slot. Switch to the correct color with the corresponding payload. Hold down power and vol up buttons. Black screen that produces the above error. Loops on this same screen.

### Crash Report

N/A atmosphere works if I use TegraGUI to inject payload.

### System Firmware Version
16.0.3

### Environment?
Atmosphère 1.5.4
fusee.bin is from the same download.

impeeza, 2023-10-11:
That's a RCM Loader problem, please use ArgonNX-SE as your payloader and you will never will need to update your RCM Loader again.
https://github.com/Storm21CH/ArgonNX-SE/releases

ELY3M, 2023-10-13:
you have to plug the rcmloader to your computer or laptop.   
see the drive and try to update all bins in all folders.      
might better to delete old bins and copy latest bins.

iCanuck, 2023-10-22:
> you have to plug the rcmloader to your computer or laptop. see the drive and try to update all bins in all folders. might better to delete old bins and copy latest bins.

This works for me when I have issues with updating the bins on the rcmloader. Sometimes it takes deleting the old files, ejecting the rcmloader then reconnect it, if the bins are still there delete them again and eject until they actually delete. Also when I copy the bins by overwriting them it never works for me.

closed 2025-04-03 by SciresM