Atmosphere-NX/Atmosphere #2169: (technically) Incorrect padding in PatchDeltaHistory

DCNick3, 2023-07-11:
AFAIK, Atmosphere uses explicit padding when dealing with structs in files/IPC. If so, the padding in [`PatchDeltaHistory`](https://github.com/Atmosphere-NX/Atmosphere/blob/8b88351cb46afab3907e395f40733854fd8de0cf/libraries/libstratosphere/include/stratosphere/ncm/ncm_content_meta_extended_data.hpp#L98) is not sufficient:

```c++
struct PatchDeltaHistory {
    PatchId source_id;
    PatchId destination_id;
    u32 source_version;
    u32 destination_version;
    u64 download_size;
    u8 reserved[4];
};
```

Due to presense of 8-byte aligned type the whole type has to be aligned by 8 bytes, but the padding at the end achieves only 4-byte alignment. Thus, the compiler will insert 4 more bytes of implicit padding.

SciresM, 2023-07-12:
I will need to check when I'm back from travel. This is a struct in a file, which means we must match Nintendo's type 1:1. If Nintendo's type has implicit padding, we will keep implicit padding. I am not unamenable to a static assert about the size, if implicit padding is correct.

SciresM, 2023-07-31:
Yeah, this is a missing-static-assert, not a wrong-type-definition.

closed 2023-07-31 by SciresM