Atmosphere-NX/Atmosphere #2170: Certain Games Crashing Right After Launching

alikendir0, 2023-07-14:
## Bug Report
### What's the issue you encountered?

Certain games such as "The Legend of Zelda Tears of The Kingdom", "Animal Crossing: New Horizons" and "Slay the Spire" crash some time after launching. So far these games are crashing that I am aware of, I haven't tested many games but some other games work just fine such as "The Longing" and "Twelve Minutes" played each of them for about 5 hours, didn't face any crashes or issues.
### How can the issue be reproduced?

Simply by launching mentioned games, depending on the game, they either crash right after launching or after playing them for about 2-3 minutes. Then switch displays the classic "The software was closed because an error occurred." screen and throws me back to the switch home menu.

### Crash Report

[TOTK Crash Report](https://gist.github.com/alikendir0/a6462afa164933c5c7630073f6b1cfa2)
[Animal Crossing Crash Report](https://gist.github.com/alikendir0/6c0ad547dfa1793f759582335dddcf15)
[Slay the Spire Crash Report](https://gist.github.com/alikendir0/8c91b29cdf680650a1075d4239b1346d)

### System Firmware Version

16.0.3

### Environment?

- Using fusèe through hekate to boot
- Using the official release of Atmosphère 1.5.4
- Homebrew software installed: Awoo Installer, DBI, Tinfoil, Daybreak, Haze, Reboot to Payload.
- Using partition-based EmuMMC with fat32 type micro SD

### Additional context?

- I am using the latest sigpatches and hekate.
- I am using a hwfly core.
- I am not using any kind of themes.
- I don't think this is a ROM issue, they work fine on other devices.
- Formatted and re-installed everything many times, got the same results.
- I don't know if this has to do anything with my issue but I also can't boot to CFW sysMMC nor CFW emuMMC from hekate, they both result in infinite black screen, only fusèe works which also from time to time results in black screen.

Masamune3210, 2023-07-14:
Piracy is not supported

Inforch, 2023-07-14:
Try install 4efir on your console, that help me to fix. Scream broke something on his atmo

Masamune3210, 2023-07-14:
What part of "not supported" was exactly lost in transmission....

SciresM, 2023-07-14:
Piracy's not supported. Also, sounds like some of your ram is fucked.

closed 2023-07-14 by SciresM