Atmosphere-NX/Atmosphere #2173: “A fatal error occurred when running Atmosphere”

BetterThanYo, 2023-07-20:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ The issue is, i tried to mod my switch for the first time, after following all the steps, i injected the payload into my switch and booted it up but it right when to a black screen saying 
“A fatal error occurred when running Atmosphere
Program ID: 010041544d530000
Error Desc: std : :abort( ) called (0xffe)”. ]</br>
[ N/A ]</br>
[ N/A ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ N/A ]</br>

### Crash Report

[ N/A ]</br>
[ N/A ]</br>
[ N/A ]</br>

### System Firmware Version

X.X.X</br>
[ I can’t check the firmware version on my switch, it’s stuck on this black screen. ]</br>
[ N/A ]</br>
[ N/A ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:fusèe
- Official release or unofficial build: Official
  - [ Idk ]
  - [ N/A ] 
     - [ N/A ]
     - [ N/A ]
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: [ Also idk ]
- EmuMMC or SysNAND: Idk
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Idk alot about this, i followed a video made 3 weeks ago by TheUndefinedGhost on youtube, how do i fix this please, it was supposed to be a gift to my gf ]

SciresM, 2023-07-20:
You didn't really fill out the template. When you get the error, you should see a message saying a report was saved to a specific file. Please zip that file and upload it, or else you cannot be helped.

BetterThanYo, 2023-07-20:
how do i upload the file, its a .bin file

SciresM, 2023-07-20:
hence "zip that file". I don't care what you do, you need to make me able to download it.

BetterThanYo, 2023-07-20:
[report_0000000005a94661.zip](https://github.com/Atmosphere-NX/Atmosphere/files/12101921/report_0000000005a94661.zip)

SciresM, 2023-07-20:
Run the archive bit fixer tool in hekate.

closed 2023-07-20 by SciresM