Atmosphere-NX/Atmosphere #2175: Pikmin 4 Stuttering

explosiveshrooms, 2023-07-24:
## Bug Report

### What's the issue you encountered?

Pikmin 4 full game stutters badly when playing. Some areas are worse than others.</br>

### How can the issue be reproduced?

For me, the issue is always present in Pikmin 4
Have tried installing as XCI, and as NSP. Tried installing using a few different options (Tinfoil, Goldleaf, and one other) but the problem is always the same

### Crash Report

No crashing is happening

### System Firmware Version

16.0.3

### Environment?

- EmuMMC or SysNAND:
  - Partition-based EmuMMC

### Additional context?

- Additional info about your environment:
Some discussion here suggests it is only happening on EmuNAND, but one user says he is seeing no stuttering on a Switch v1 on Emunand, but yes stuttering on a v2 switch on Emunand - https://www.reddit.com/r/SwitchPirates/comments/153quix/pikmin_4_massive_stuttering_not_present_in_demo/

Some discussion here too but it is less useful - https://www.reddit.com/r/SwitchPirates/comments/153g9uf/pikmin_4_stutterlag_anyone_else_having_this/

I have not yet tested the game in sysnand myself

explosiveshrooms, 2023-07-24:
The reason i specified "Pikmin 4 full game" is because some people have said the issue isn't present on the Pikmin 4 Demo

zemixe, 2023-07-24:
Piracy is not supported.

explosiveshrooms, 2023-07-24:
> 

thanks, i didn't pirate it but i removed a couple of links that may have been against anti-piracy rules

SciresM, 2023-07-24:
Piracy is not supported. You did in fact pirate it, per `tried installing as XCI, and as NSP`.

Anyway, I've been playing Pikmin 4 a lot, and have like 30 hours in the game using Atmosphere.

No stuttering problems whatsoever.
It's gonna be an issue with your setup, not an atmosphere issue.

closed 2023-07-24 by SciresM