Atmosphere-NX/Atmosphere #2176: Can't load CFW or OFW Error 0xff0 Title ID 524850534f53454d

madmaxen9, 2023-07-28:
Was loading a game onto the switch via DBI installer and the screen went yellow. Every load after this it either hangs up on the nintendo switch screen or goes to the Yellow writing with the subject error. 
Yes, updated to 16.0.3 yesterday and prior to install of game it was working on the new version. 
Upgraded to 16.0.3

Attempting to load OFW (AutoRCM off) leads to getting stuck on Nintendo Switch load screen
Attempting to load Sysnand/EMUMMC from Hekate ends in error screen and sometimes stuck on Nintendo Switch screen. 

### Crash Report

No crash report, just fatal error but not sure if this shows anything (https://gist.github.com/madmaxen9/4f8a8ec78f7f91730af62ccb50761345)

### System Firmware Version

16.0.3 I believe, unable to boot into home screen to check

### Environment?

Official Hekate 6.0.5
- Official release or unofficial build: 
- Homebrew software installed: Atmosphere
- EmuMMC or SysNAND: both

### Additional context?

No crash reports generated on this date, get fatal error reports but they do not appear to show much

SciresM, 2023-07-28:
Piracy is not supported.

closed 2023-07-28 by SciresM

madmaxen9, 2023-07-28:
> Piracy is not supported.

Apologies, did not mean to reference any. Atmosphere is my issue to get sysnand and EMU to load

madmaxen9, 2023-07-28:
[image]

SciresM, 2023-07-28:
"Piracy is not supported" means piracy is not supported, sorry.