Atmosphere-NX/Atmosphere #2177: Atmosphere Crash in Sys and Emu

madmaxen9, 2023-07-28:
## Bug Report

Can't load CFW or OFW Error 0xff0 Title ID 524850534f53454d

### What's the issue you encountered?

Atmosphere is crashing, screenshot of issue here. Unable to load sys or emu
[image]

### How can the issue be reproduced?

Loading into Hekate 6.0.5 and boots to Atmosphere screen, then Nintendo and either get the fatal error message or freezes at NS logo

### Crash Report

(https://gist.github.com/madmaxen9/4f8a8ec78f7f91730af62ccb50761345)

### System Firmware Version

X.X.X</br>
16.0.3

### Environment?

Off Hekate 6.0.5
- Do you have additional kips or sysmodules you're loading: N
- Homebrew software installed: N/A
- EmuMMC or SysNAND: Both

### Additional context?

- Additional info about your environment:
Contents folder screenshot
[image]

SciresM, 2023-07-28:
Piracy is not supported. I just closed your other issue.

closed 2023-07-28 by SciresM

madmaxen9, 2023-07-28:
I'm sorry but where did I say anything about piracy, I'm simply trying to solve my Atmosphere crash issues

SciresM, 2023-07-28:
`Was loading a game onto the switch via DBI installer`. As I said, piracy is not supported.