Atmosphere-NX/Atmosphere #2179: Fatal Error Problem With Program ID 010041544d530000 Whenever I Play My Modded Super Smash Bros. Ultimate

Rosacakes, 2023-08-02:
## Bug Report

### What's the issue I encountered?

Whenever I try to play my modded version of Super Smash Bros. Ultimate, I could play it for a little bit before it crashes and a fatal error shows up. Before, I've been able to play this modded Smash flawlessly until one day these errors started popping up.</br>
I installed the ARCropolis plugin and my mods, most of which are skins.</br>

### How can the issue be reproduced?

Boot into RCM mode, launch Hekate, boot into Atmosphère, launch modded Smash Ultimate, play for a little bit until the fatal error screen pops up. The time I could play Smash before the fatal error shows up ranges from 10 seconds to 10 minutes.</br>

### Crash Report

It's a fatal error with a program ID of 010041544d530000. I don't know what program it is.</br>

Here's all the reports I got: [fatal_errors.zip](https://github.com/Atmosphere-NX/Atmosphere/files/12238102/fatal_errors.zip)

### System Firmware Version

16.0.3</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
   - Hekate
- Official release or unofficial build: 
  - Official release Version 1.5.4
- Homebrew software installed: hbmenu, Tinfoil, Checkpoint, Daybreak
- File-based EmuMMC

### Additional context?

- This only happens to Super Smash Bros. Ultimate. I just want to play my modded version of Smash with no problems so I could have some good laughs with my friends.

SciresM, 2023-08-02:
fs::ResultPortSdCardDataCrcError,
fs::ResultPortSdCardSdCardNotReadyToVoltageSwitch,
fs::ResultPortSdCardTuningFailed,

These are all SD card hardawre errors. You should try a new SD card, and if that doesn't work it might be an issue with your connector.

Either way, this is a hardware problem, not an atmosphere problem.

closed 2023-08-02 by SciresM

Masamune3210, 2023-08-02:
Piracy is not supported.