Atmosphere-NX/Atmosphere #2180: Can't Longer to "Turn Off" My Switch Anymore on AMS 1.5.4!

Mr-Denni, 2023-08-02:
**What's the issue you encountered?**
I got problem (which just I recently notice) after upgrade to AMS 1.5.4 FW 16.0.3.
When I tried to power off the Switch from HOME Menu (by pressing POWER button for 3s & choose "Turn Off"), it only power off for several seconds (20s maybe), then the Switch BACK ON again??!
It happened not just only on CFW mode (SYSNAND), but also on stock OFW mode too!
Worst, the POWER button also behave the same if I press it for about 12s when usually it power off the Switch completely!

**How can the issue be reproduced?**
On HOME Menu, press the POWER button for 3s and choose "Turn Off", then wait for about 20s and voila!
Or press the POWER button for 12s until the Switch turn off (blank screen), then wait for about 20s.

**Crash Report**
No crashes to report

**System Firmware Version**
16.0.3 | AMS 1.5.4|S

**Environment?**
Hekate 6.0.5
Official (bootloader and atmosphere from github)
SYSNAND

**Additional context?**
I notice the issue first after installing 16.0.3 with AMS 1.5.3 release.
Continues on 1.5.4.

SciresM, 2023-08-02:
This is going to be some issue with your setup, not an issue with atmosphere.

That said...you can presumably inject hekate and power off through its menu, or hold power button for 30 seconds.

closed 2023-08-02 by SciresM

Masamune3210, 2023-08-02:
Makes me wonder if they have AutoRCM enabled

Mr-Denni, 2023-08-02:
> This is going to be some issue with your setup,...

Base on this info, I dug it for more info in the net and found a solution by changed 2 parameters in "hekate_ipl.ini":
autohosoff=2 (iniatially 0)
autonogc=0 (iniatially 1)
Problem solved!

masagrator, 2023-08-02:
autonogc is for cartridge slot fw, completely unrelated to your issue.