Atmosphere-NX/Atmosphere #2181: Error while booting after updating

Ishidad, 2023-08-03:
## Bug Report

### What's the issue you encountered?

I updated my console from 14.0.X to 16.0.3
Updated both Atmosphere and Hekate
Updated the payload injector
When booting I get this error message

A fatal error occurred when running Atmosphere.
Program ID: 010000000000bd00
Error Desc: std::abort() called (0xffe)

Report Saved to...
Press power to reboot

### How can the issue be reproduced?

booting from fusee

### Crash Report

[report_000000000af39401.zip](https://github.com/Atmosphere-NX/Atmosphere/files/12253228/report_000000000af39401.zip)

### System Firmware Version

16.0.3</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: atmosphere-1.5.4-master-3cb54e2b4
- Do you have additional kips or sysmodules you're loading: no that I know
- Homebrew software installed: Don't remember all, goldleaf and a few more I never use
- EmuMMC or SysNAND:
  - N/A

### Additional context?

N/A

masagrator, 2023-08-03:
> * no that I know

You could really to spend on this few more minutes and you would understand the issue.
You have outdated MissionControl Installed. Delete folder "010000000000bd00" from contents since it seems you don't use it.

Ishidad, 2023-08-05:
Thanks for the help, I did that and now I almost got to boot, now I'm getting this error 
[image]
I'm looking it up as I creat the post

Ishidad, 2023-08-05:
I found that is related to overlays, I had Retro reloader overlay, how can I remove it?

Ishidad, 2023-08-05:
Had to remove a bunch of folders from content and after that all is working, thanks for the help

Ishidad, 2023-08-05:
Sorted

closed 2023-08-05 by Ishidad

kohdenny, 2024-04-17:
> > * no that I know
> 
> You could really try to spend on this few more minutes and you would understand the issue. You have outdated MissionControl Installed. Delete folder "010000000000bd00" from contents since it seems you don't use it.

Thx bro, my switch have this problem & after delete the folder, the atmosphere run normal again.

DarnellTyrese, 2024-08-30:
> > * no that I know
> 
> You could really try to spend on this few more minutes and you would understand the issue. You have outdated MissionControl Installed. Delete folder "010000000000bd00" from contents since it seems you don't use it.

I have the same problem, but the folder does not exist. What should I do?