Atmosphere-NX/Atmosphere #2184: Blackscreen after launching fusee.bin

BobbyDig, 2023-08-14:
## Bug Report

### What's the issue you encountered?

I'm having a blackscreen after booting from fusee.bin
I can start hekate without any problems, but when I want to launch I get the Atmosphere Logo and then I get a invinite blackscreen.

### How can the issue be reproduced?

I have no idea how to reproduce.
I mean not long ago I had the same issue and started from beginning.
I thought it would be a format thing, but now it's happening again and I'm confused.
In my case it worked for a week or two and when I'm restarting the switch then I get a blackscreen now.
Maybe there are to many games on or tinfoil doesn't install games correctly. I have no idea.

### Crash Report

Unluckly there is no crashreport.
There are only erpt.reports.

### System Firmware Version

16.0.3</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - 16.0.3 Pack from Better Gaming Discord
   - hekate: hekate_ctcaer_6.0.4.bin
   - atmosphere: atmosphere 1.5.3
     (but I also tried it with official release)
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: 
  - tinfoil, SimpleModManager, linkkalho, appstore, daybreak, haze
- EmuMMC or SysNAND:
  - EmuMMC partition-based

### Additional context?

- Additional info about your environment:
- No I don't think so.

BobbyDig, 2023-08-14:
Here are the erpt.reports
[erpt_reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/12333331/erpt_reports.zip)

Azuuuma, 2023-08-22:
I encountered the same issue. After reinstalling the virtual system, I found an infinite black screen. I used hekate 6.0.6 and Atmosphère 1.5.5 versions.

closed 2023-08-28 by SciresM