Atmosphere-NX/Atmosphere #2185: Could not start the software for several games ID : 0100D1B006744000 + 01005FF002E2A000 +010077B00E046000

mathieuv56, 2023-08-14:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Games works well on 1.3.2 but failed after an upgrade on 1.5.3</br>
No change</br>
</br>

### How can the issue be reproduced?

[ * ]</br>
Upgrade from AMS 1.3.2 to 1.5.4 (+ hekate + fusee + sigpatches 1.5.4) and from the previous firmware to 16.0.3 then try to execute one of the mentionned games
You will reach the error message "could not start the software"</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
None</br>

### System Firmware Version

16.0.3</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - 1.5.4
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - Don't know
  - 
### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

mathieuv56, 2023-08-14:
Other games are working fine

SciresM, 2023-08-14:
Piracy is not supported.

closed 2023-08-14 by SciresM