Atmosphere-NX/Atmosphere #2186: Add an option to set different theme between EmuNAND and SysNAND

leElvyn, 2023-08-15:
## Feature Request

### What feature are you suggesting?
#### Overview:

Allow for clear and obvious differentiation between running an EmuNAND and SysNAND by letting the user set two different themes, for both modes.

#### Smaller Details: 

A simple way to implement this is to first try to pull the theme as it is currently done. If this fails because no theme has been set in the default folder, atmosphere will try to pull from an "emunand" folder if CFW is running on EmuNAND, or from a "sysnand" folder if CFW is running on SysNAND. This preserves all current setup, makes the feature completely optional, and easy to use and implement.

#### Nature of Request:  

- Change
  - Change the way themes are loaded when starting CFW

### What component do you feel this would best fit within?
- [**Thermosphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's emuNAND implementation.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?

EmuNAND is an extremely powerful feature allowing a user to do any action and experiment with potentially harmful actions, without taking the risk of having to restore a system backup, or getting banned. However, it is easy to accidentally mistake the SysNAND for EmuNAND, and vice versa, which can result in ban, or permanent damage simply because there are no way to distinguish between the two modes at a glance, other than using the default theme.

SciresM, 2023-08-15:
This works with built in light/dark theme. Just noting that, since people *do* use that, it's only the unofficial "themes"...which they could also tackle in software, I'd expect, though I guess I don't actually expect they ever will.

But anyway, this boils down to being a duplicate of #704. The long-standing answer is that it's really, really complicated, and there are trade-offs that aren't obviously worth making. More doable now that we have stratosphere.romfs, I guess.

closed 2023-08-15 by SciresM

leElvyn, 2023-08-15:
I see, thanks

Is there at least a way to enable a big scary warning on the homescreen of SysNAND ?