Atmosphere-NX/Atmosphere #2193: IPS patch alternative beyond the 24-bit address limit

mrdude2478, 2023-09-14:
## Feature Request:
IPS patch alternative beyond the 24-bit address limit.

### What feature are you suggesting?
Increased IPS Patch address patching.

#### Overview:
Currenly IPS patch addresses are limited to 24 bit addresses, if for example to patch a file at an address higher that this address we cannot do this with an IPS patch and need to use a program such as SaltyNX instead. Or patch the file and place it on the sd card in the contents folder.

#### Nature of Request:  
A new format of IPS patch (but similar) that allows to patch higher than the 24bit address limit.

### What component do you feel this would best fit within?
Not sure which part of Atmosphere deals with IPS patching.

### Why would this feature be useful?
It can be used to create small patch files to patch out checks for internet connections, game mods etc. This elimates the need for extra program to be installed which might not work in future updates, and eliminates to need to install large files to the sd card.

SciresM, 2023-09-14:
Atmosphere already supports IPS32.

closed 2023-09-14 by SciresM

SciresM, 2023-09-14:
IPS32 is just IPS, with the following difference:
* Header is IPS32 instead of PATCH
* Footer is EEOF instead of EOF
* Offset is 4 bytes instead of 3.

Consider e.g. using https://github.com/leoetlino/sips

mrdude2478, 2023-09-14:
Thanks. I never realised this was a thing, it's difficult to find info about it with google. Testing now though, A+++