Atmosphere-NX/Atmosphere #2194: Switch not booting, screen just turns black and nothing happens

Wachitozs, 2023-09-21:
After updating to this new atmosphere update, when i start my switch by TegraRCM and Hekate, when clicking launch it just loads the atmosphere logo and then my screen turns black and stays there and nothing happens. Ill leave a video for a better look, wonder if anyone can help me pls. I used this switch pack from a spanish youtuber https://mega.nz/folder/DMYXSAAA#r_yikDIsE0Ha1IJxnw2hNg where it includes the SD pack and also the TegraRCM and Hekate pack. Im new to Github and let me know if im doing something wrong. Heres the video of the problem im having: https://imgur.com/a/frBTONm

Masamune3210, 2023-09-21:
A. Don't use packs, they are routinely crap and include things you don't need like sysmodules you will never use.
B. Please go back and actually fill out the issue template instead of completely blowing it away like you did.

SciresM, 2023-09-21:
You didn't use the issue template.

closed 2023-09-21 by SciresM