Atmosphere-NX/Atmosphere #2195: Viewing Intellectual Property Notices for Pokémon HOME crashes system

DevanWolf, 2023-09-25:
## Bug Report
### What's the issue you encountered?
When viewing Intellectual Property Notices for the title Pokémon HOME it crashes Atmosphère instead opening the document. Viewing Intellectual Property Notices for other titles work fine. This did not happen in stock firmware.
### How can the issue be reproduced?
1. Open Details for Pokémon HOME in HOME Menu
2. Select Intellectual Property Notices
### Fatal Report
```Atmosphère Fatal Report (v1.1):
Result:                          0x2EEA02 (2002-6005)

Program ID:                      010000000000100f
Process Name:                    LibAppletOf
Firmware:                        16.1.0 (Atmosphère 1.5.5-master-4fe9a89ab)
General Purpose Registers:
         FP:                     00000043eb3fedf0
         LR:                     0000003d5fc7aed4
         SP:                     00000043eb3fedf0
         PC:                     0000003d5fc932c0
Start Address:                   0000003d5f600000
Stack Trace:
        ReturnAddress[00]:       0000003d5f9f0ca8
        ReturnAddress[01]:       0000003d5f9f06a0
        ReturnAddress[02]:       0000003d5f9f07b8
        ReturnAddress[03]:       0000003d5f8a37ac
        ReturnAddress[04]:       0000003d5f8a27fc
        ReturnAddress[05]:       0000003d5f8a28b8
        ReturnAddress[06]:       0000003d5f8a2998
        ReturnAddress[07]:       0000003d5f9115a8
        ReturnAddress[08]:       0000003d5fc531b0
        ReturnAddress[09]:       0000003d5f787dd4
        ReturnAddress[10]:       0000003d5f7866f8
        ReturnAddress[11]:       0000003d5f76f828
        ReturnAddress[12]:       0000003d5f60560c
        ReturnAddress[13]:       0000003d5f604778
        ReturnAddress[14]:       0000003d5f604488
        ReturnAddress[15]:       0000003d5f82c2d0
        ReturnAddress[16]:       0000003d5f9f0504
        ReturnAddress[17]:       0000003d5f6000d4
Stack Dump:                               00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             0043eb3fedf0 90 ee 3f eb 43 00 00 00 a8 0c 9f 5f 3d 00 00 00
                             0043eb3fee00 01 01 00 00 00 00 10 00 0c 00 00 80 01 00 00 00
                             0043eb3fee10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0043eb3fee20 05 00 00 00 00 00 00 00 00 00 00 00 08 00 00 00
                             0043eb3fee30 00 00 00 00 14 00 00 00 00 00 60 5f 3d 00 00 00
                             0043eb3fee40 00 20 00 00 00 00 00 00 03 00 00 00 00 00 00 00
                             0043eb3fee50 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0043eb3fee60 78 ee 3f eb 43 00 00 00 15 00 00 00 00 00 00 00
                             0043eb3fee70 00 49 17 60 3d 00 00 00 50 02 00 00 00 00 00 00
                             0043eb3fee80 00 ee 3f eb 43 00 00 00 00 c2 c9 54 57 00 00 00
                             0043eb3fee90 00 ef 3f eb 43 00 00 00 a0 06 9f 5f 3d 00 00 00
                             0043eb3feea0 52 2e e3 5f 3d 00 00 00 f0 f0 3f eb 43 00 00 00
                             0043eb3feeb0 52 2e e3 5f 3d 00 00 00 00 00 00 00 00 00 00 00
                             0043eb3feec0 a0 16 09 60 3d 00 00 00 b0 17 09 60 3d 00 00 00
                             0043eb3feed0 38 ef 3f eb 43 00 00 00 00 00 00 00 00 00 00 00
                             0043eb3feee0 e0 16 09 60 3d 00 00 00 4c 3e e9 5f 3d 00 00 00
TLS Address:                 0000005754c9c200
TLS Dump:                                 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             005754c9c200 06 00 10 00 0c 00 00 80 01 00 00 00 f2 01 08 00
                             005754c9c210 00 00 00 00 50 02 00 00 00 49 17 60 0c 00 00 d0
                             005754c9c220 53 46 43 49 01 00 00 00 02 00 00 00 00 00 00 00
                             005754c9c230 02 ea 2e 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c240 b8 12 00 00 00 00 00 00 5c 00 00 00 00 00 00 00
                             005754c9c250 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c260 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c270 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c280 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c290 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c2a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c2b0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c2c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c2d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c2e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005754c9c2f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
### System Firmware Version
16.1.0
### Environment?
- Bootloader Atmosphère launched by: fusèe via hekate
- Official release version: 1.5.5
- Do you have additional kips or sysmodules you're loading: sys-clk
- Homebrew software installed: N/A
- SysNAND

SciresM, 2023-09-25:
Can you please post an image of your atmosphere/contents directory?

fs::ResultInvalidPathFormat seems...very unusual, for this.

SciresM, 2023-09-25:
I cannot reproduce this. I am successfully able to open all three of support information, important information, and intellectual property notices for Pokemon Home 3.1.0.

DevanWolf, 2023-09-26:
Edited issue, I realized that software corruption is what caused it. After deleting all folders in Nintendo/registered and reinstalled Pokémon HOME, it works fine.

closed 2023-10-12 by SciresM