Atmosphere-NX/Atmosphere #2196: A fatal error occurred when running Atmosphere - Program ID: 010041544d530000

BtotheBen, 2023-10-05:
## Bug Report
</br>

### What's the issue you encountered?
</br>
When booting into CFW, I get the error message. Beforehand I updated atmosphere from 1.5.4 to 1.5.5 and CFW and OFW to 16.1.0. When searching for the issue online I tried Fix Archive Bit, but it did not fix the issue.

Error Message:
A fatal error occurred when running Atmosphere.
Program ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

Report saved to /atmosphere/fatal_errors/report_0000000001117e08d.bin
</br>

### How can the issue be reproduced?
</br>
Update Atmosphere from 1.5.4 to 1.5.5
Update Firmware OFW to 16.1.0
Boot into CFW

### Crash Report

[report_000000000117e08d.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/12822021/report_000000000117e08d.bin.zip)

### System Firmware Version

16.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: I used hekate
- Official release or unofficial build: Official Release 1.5.5
- Do you have additional kips or sysmodules you're loading: no idea
- Homebrew software installed: no idea
- EmuMMC or SysNAND: I use a partition-based EmuMMC

### Additional context?

- Additional info about your environment:
- I have an OLED Switch

SciresM, 2023-10-05:
The problem is that you updated to OS version 16.1.0 but didn't update atmosphere properly. This report is from atmosphere 1.5.4, and the error it describes is "the official firmware version is higher than supported", which makes sense because 16.1.0 requires atmosphere 1.5.5.

Make sure that you don't have anything in atmosphere/kips or anything, and re-extract the latest release zip to the sd card.

closed 2023-10-05 by SciresM

BtotheBen, 2023-10-05:
Thank you for the quick answer!